OrcaSlicer/OrcaSlicer #4514: Flashforge Adventurer 3 Profile - Temp and Fan Control

JediMasterBob, 2024-03-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Flashforge Adventurer 3 Pro 2

### How to reproduce

Issue is via the standard G-Code output. Is reproducible on every output.

Temperature Control: temperature is not being controlled. The printer firmware has an issue with comments (;) on the M104 lines. If there is a comment at the end of this line, the entire line is ignored and temperature is not changed.

Fan Control: Adventurer 3 seems to have a very rudimentary fan control. It is either ON or OFF, but nothing in between. You can check how FlashPrint generates fan control, using only M106 (fan on) and M107 (fan off). I see that Orca tries to use "M106 S0" to set 0% fan speed, but this fails as Adventurer 3 ignores the parameter. These "M106 S0" commands should be replaced by "M107".

Both causes of the issues were discovered by: @rusty (https://github.com/RustyRaindeer)

topic discussions
Temperature issue: https://github.com/RikshaDriver/adventurer3-orcaslicer-profile/issues/1
Fan issue: https://github.com/RikshaDriver/adventurer3-orcaslicer-profile/issues/2

### Actual results

Temperature is not correctly controlled/changed at the specified layers. Fan control is not properly controlled. 

### Expected results

Temp and fan should be properly controlled during print. 

### Project file & Debug log uploads

[temperature_tower (2).txt](https://github.com/SoftFever/OrcaSlicer/files/14623662/temperature_tower.2.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2024-03-17:
It's doesn't look like a Orca issue.
Can you raise ticket in their system?

closed 2024-03-17 by SoftFever

Pseudopay, 2024-07-04:
The Adventurer 3 controls the fan according to temperature and ignores the g-code, which is normal for this printer.

dbaarda, 2025-09-01:
And a workaround;

I use the following custom machine gcode and post-processing script to make the Orca output Adv3 compatible;

* https://github.com/dbaarda/workspace-tools/blob/main/3DPrinting/Adv3_machine_start.gcode
* https://github.com/dbaarda/workspace-tools/blob/main/3DPrinting/Adv3_machine_end.gcode
* https://github.com/dbaarda/workspace-tools/blob/main/3DPrinting/Adv3_layer_change.gcode
* https://github.com/dbaarda/workspace-tools/blob/main/3DPrinting/toadv3.sh

This transforms the gcode output so that Oca's filament cooling settings for the extruder fan work, and the enclosure fan is controlled by the "Exhaust Fan" settings. The "Auxilery part cooling fan" settings are not used, and I modify the printer config to disable it. These work with the following caveats;

1. The Av3 extruder fan is only "on" or "off" (though Flashforge claim "on" means automatic depending on the temp). Any `M106 S<0-255>` fan speed commands emitted by Orca for speeds less than 99 are transformed into `M107` "off" commands, and anything more into `M106` "on" commands.
2. The Adv3 doesn't push fan on/off cmds onto the execution queue, but seems to execute them directly as the gcode is parsed, so they execute before the other gcode commands before them. This means it cannot do the fine-grained fan-on/off for bridges, overhangs, etc actions that Orca can generate, because the fan will turn on and off again well before the bridge/overhang is printed. At best it can be used for turning the fan on/off for particular layers, and even then you will notice the fan turning on/off prematurely.
3. The Adv3 enclosure fan does seem to understand fan-speed settings (flashprint output includes the speed argument), but I haven't really tested it. Any `M106 P3 S<1-255>` exhaust fan speed commands emitted by Orca are transformed into `M651 S<1-255>` Adv3 enclosure fan speed commands, and `M106 P3 S0` is transformed into `M652` Adv3 enclosure fan off commands.
4. The `Adv3_machine_start.gcode`, `Adv3_machine_end.gcode` and `Adv3_layer_change.gcode` printer config fragments include redundant fan control commands sufficient to set the fans to turn on at the desired layer and off at the end. This means they should be sufficient to drive the fans without using the post-processing script.
5. The `toadv3.sh` post-processing script is a bash/sed script, so it might not work on non-linux systems. If people really want I could probably easily translate it into Python, but bash/sed is much more lightweight. 
7. The post-processing script does a few more things than just transform fan commands. It also strips trailing comments (Adv3 ignores commands with trailing comments), strips `M106 P2 ...` Auxiliary fan commands (Adv3 has no auxiliary fan), strips `.0` of the end of temperature settings (Adv3 doesn't like those, and though Orca currently doesn't seem to add them, it has caused problems in the past), and transforms `G0` into `G1` (Adv3 doesn't understand `G0` cmds).

Also note that the V2.3.1-alpha builds includes the following fixes that make printing direct from Orca work;

1. The default printer machine_start.gcode nolonger puts the pre-extrude out of the print area (this fix is also included and improved in my `Adv3_machine_start.gcode`. See #6282 for details.
2. The network connection code includes an extra delay after sending the file and the `~M29` saveFileCommand. See #8905 for details.