OrcaSlicer/OrcaSlicer #4521: Klipper Accel to deccel already obsolete

DVSVIDEO, 2024-03-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Recently in the Klipper firmware, the acceleration to deceleration function was removed and the minimum cruise ratio was introduced instead.
[image]

In OrcaSlicer, the old value is still shown as %.
[image]

Please change it to the current value.
Thanks in advance!

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Update the function

SoftFever, 2024-03-17:
Please check the discussion here(you can scroll the bottom to save some time): https://github.com/Klipper3d/klipper/pull/6418

I will leave this feature there for a few months as it's working for both old and new Klipper versions as expected.

When most people have migrated to the new Klipper version, I plan to remove these options and add an option in the printer settings, as we don't need to adjust the new minimum_cruise_ratio per extrusion role.

We probably don't need to set it in slicer side anymore as I think it's should be part of the Klipper config considering how the new parameter works.

vgdh, 2024-03-17:
same https://github.com/SoftFever/OrcaSlicer/issues/3084

SoftFever, 2024-03-17:
Duplicated

closed 2024-03-17 by SoftFever