OrcaSlicer/OrcaSlicer #4523: Flow display in color scheme goes haywire

Rickthebig, 2024-03-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0 beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Customized printer (DIY), based on Troodon template (RRF)

### How to reproduce

1. I just slice a model.
2. Look at the Volumetric flow rate display and notice the exorbitant range.
[image]

3. Look for extremely high retractions followed by extrusions, mostly same amount (on a limited number of places).
[image]
4. Scanning layer by layer, other places with abnormal retraction/extrusion behaviour can be identified:
[image]

### Actual results

Print the part and observe  filament the unusual long moves. The part will be either overextruded or underextruded in particular places where the high retractions/extrusions occur.

### Expected results

A number of gcode lines have wrong E values. When manually corrected (by replacing those values) the part will print as expected.

### Project file & Debug log uploads

[Cooler-Test.zip](https://github.com/SoftFever/OrcaSlicer/files/14625273/Cooler-Test.zip)
[debug_Fri_Mar_15_16_13_12_18400.log](https://github.com/SoftFever/OrcaSlicer/files/14625274/debug_Fri_Mar_15_16_13_12_18400.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]
[image]

[image]

vgdh, 2024-03-17:
Similar issue
I have flow constrain at 8mm3 but at the same time soft show me higher flow color scheme values that are useless.
[image]

vgdh, 2024-03-17:
Similar issue
I have a flow constraint of 8mm^3, but simultaneously, the software shows me higher flow color scheme values that are useless.
[image]

SoftFever, 2024-03-17:
Seems the color display is broken for absolute extrusion mode.

SoftFever, 2024-03-17:
> Similar issue I have a flow constraint of 8mm^3, but simultaneously, the software shows me higher flow color scheme values that are useless. ![Снимок](https://private-user-images.githubusercontent.com/15322782/313453798-c3d9e514-3eeb-4fc0-a789-46966361e977.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA2NzkyMzEsIm5iZiI6MTcxMDY3ODkzMSwicGF0aCI6Ii8xNTMyMjc4Mi8zMTM0NTM3OTgtYzNkOWU1MTQtM2VlYi00ZmMwLWE3ODktNDY5NjYzNjFlOTc3LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE3VDEyMzUzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTczZTNkZjQ2MDYxZTUwNjE5ZDVkZWY5Y2RkYzYyNDA5NzY1MjllOGRlZGQyNTI1Mzk4NjE5NjQ3NDYxZDNjMTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.9484818ekqB4sWhDC6d2imUHdj5OoqgRZSfB9QFYj7s)

It's a different issue.
When using Arachne, flow will be slightly off where the extrusion width changes

Rickthebig, 2024-03-17:
> Seems the color display is broken for absolute extrusion mode.

Perhaps I did not explain clearly.
At first, I also though that this is a display problem. It is not.
The volumetric extrusion range in fact comprises all existing flow rates as sliced, but they are sliced wrong. At some particular spots, something weird occurs and the filament is pulled or pushed (or both) several cm. This can be seen as a disruption or discontinuity of the extrusion from the E numbers in the gcode file. If those entries are corrected/removed manually, and the gcode file reloaded, a correct flow range will be shown (and the piece would print nice).

So this is a real problem, not just a display problem. But it happens only at certain conditions, which I am trying to identify. If switching to an unmodified printer setting, then extrusion will be alreight, and the color scheme too.

Rickthebig, 2024-03-17:
OK, I found the setting that confuses the slicer.
In the printer definition-> Basic Information->Advance:
When "Use relative E distances" is not checked, the problem occurs.
Using absolute E distances does not work clean in this version!
**Cause found, problem unsolved.**

SoftFever, 2024-03-18:
> Perhaps I did not explain clearly. At first, I also though that this is a display problem. It is not. The volumetric extrusion range in fact comprises all existing flow rates as sliced, but they are [...]

that's retractions. due to the display bug, you don't see the dots that representing retractions.
Have you printed it by any chance?

closed 2024-06-26 by github-actions[bot]