OrcaSlicer/OrcaSlicer #4524: Orca sliced files crash Flashforge Adventurer 5M

ad5muser, 2024-03-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 7

### Printer

Flashforge Adventurer 5M

### How to reproduce

Slice any file in Orca Slicer 2.0.0 or 1.9.1 and save the gcode file on a USB drive. Insert the USB drive into Adventurer 5M and select the print menu - at this point the printer will become unresponsive and will need to be powered off.

### Actual results

When a USB is inserted into the printer with a gcode file sliced by Orca 2.0.0 or 1.9.1 and when the print menu is selected on the printer the printer will become unresponsive and need to be powered off. This only occurs with files sliced by recent versions of Orca: files sliced with FlashPrint, Cura and earlier versions of Orca work fine.

### Expected results

After the above steps, the printer should remain responsive and you should be able to navigate to the gcode file you want to print and print it. Gcode files sliced by FlashPrint, Cura and earlier versions or Orca work correctly.

### Project file & Debug log uploads

[Guard.zip](https://github.com/SoftFever/OrcaSlicer/files/14625690/Guard.zip)
[crash_Sun_Mar_17_10_05_48_0.log](https://github.com/SoftFever/OrcaSlicer/files/14625699/crash_Sun_Mar_17_10_05_48_0.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Orca Slicer happened to crash during the most recent test, and I attached the log for completeness and because it was asked for. But I don't believe it is related to the main issue, which is that the sliced files seem to be incompatible with Flashforge Adventurer 5M.

SoftFever, 2024-03-17:
Can you raise support ticket in their system?
I'm not able to support specific printer model

closed 2024-03-17 by SoftFever

ad5muser, 2024-03-22:
> Can you raise support ticket in their system? I'm not able to support specific printer model

Hi Softfever, thanks for your reply.
Yes I raised a ticket with Flashforge first before raising it here. The opinion of Flashforge Service was because the printer worked with all other files, including files sliced with Orca V 1.7.0 and Cura, it was believed to be most likely that the problem was with Orca Slicer V 1.9.1 and 2.0.0.
I would be grateful if you could share your thoughts on this or look into it.

Thanks
Andrew

Symbiose1234, 2024-04-18:
Hi,
I tested it yesterday with v2.1.0 and the printer also freezes :)