OrcaSlicer/OrcaSlicer #4526: Orca crashes while slicing fuzzy stuff

rmwerder, 2024-03-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 11 22000.2713

### Additional system information

AMD Ryzen 5 5600X 6-Core Processor                3.70 GHz
32,0 GB RAM
Nvidia GeForce RTX 3070

### Printer

BambuLab P1S

### How to reproduce

1. get https://www.thingiverse.com/thing:3895760
2. set Others - Special Mode - Fuzzy Skin -> Contour and Hole
3. Select "Slice Plate"

### Actual results

Orca crashes (Window is closing without message, task terminates). BambuStudio works fine with identical settings

### Expected results

it should slice the model

### Project file & Debug log uploads

[crash_Sat_Mar_16_20_32_20_1.log](https://github.com/SoftFever/OrcaSlicer/files/14626986/crash_Sat_Mar_16_20_32_20_1.log)
[crash.zip](https://github.com/SoftFever/OrcaSlicer/files/14626999/crash.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

vgdh, 2024-03-17:
works fine with latest night beta
[image]

SoftFever, 2024-03-17:
I managed to repro this bug with the attached project file.
Fixed in latest nightly build

rmwerder, 2024-03-17:
thanks for the quick reaction. Will download and test tonite. R.

closed 2026-02-19 by ianalexis