OrcaSlicer/OrcaSlicer #4530: Scarf Seam Painting

CCS86, 2024-03-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Since scarf seams work great in some situations, but quite poorly in others, controlling when to use a scarf seam, and when to use a traditional seam, appears to be an issue that won't go away. 

The conditional scarf seam settings do help somewhat, but don't leave the user with enough control.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I think that adding a feature to the seam painting function, to allow painting specifically scarf and traditional seams would give us all the control we need.

mikeq3, 2024-04-17:
I agree with that request. There seems to be no control over where the scarf seam is placed. Seam painting doesn't appear to work for a scarf seam. Having seam painting for scarf seams as well as more control over which seams are scarf would be an excellent feature.

CCS86, 2024-04-17:
> I agree with that request. There seems to be no control over where the scarf seam is placed. Seam painting doesn't appear to work for a scarf seam. Having seam painting for scarf seams as well as more control over which seams are scarf would be an excellent feature.

Seam painting does work with scarf seam positioning.

If you look at the FLOW visualization, you can see where the scarf seam begins (that's what you define with painting). 

The seam marker in gcode view, shows where the scarf seam ends. That is controlled by starting position, perimeter loop direction, and the scarf length.

mikeq3, 2024-04-17:
If I paint a straight seam on one side of the model shouldn't the displayed seam end match that line being that scarf length is fixed? The painted seam is the starting position, is it not?

CCS86, 2024-04-17:
Again, look at the flow view. It will make more sense.

mikeq3, 2024-04-17:
I see what you are saying, thanks.

Lungenpest, 2024-07-23:
would be great

closed 2024-10-29 by github-actions[bot]