OrcaSlicer/OrcaSlicer #4531: Scarf Seam fails when Seam gap is set 20%

erdo100, 2024-03-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

AMD Ryzen 5 3600 6 Core
64GB DDR4 3200

### Printer

Ender 3S1 with Klipper, d=0.4 mm nozzle

### How to reproduce

For test purpose I am printing small elliptical prisms with width:20mm, 10mm and height 10 mm.
Layer height 0.3mm
No Infill, No Bottom, No top
3 Walls with 0.45mm
Scarf Settings: 
Seam Position Back
Seam Gap 20%
Rest is default (Scarf length 20 mm, Steps 10,)

### Actual results

Printer makes prints the wall, but instead of making the scarf, lifts and then makes an additional loop to make the scarf. I guess it has lost the start point of the scarf and prints it from wrong direction. This leads to 2 big seams.

Video during printing:
https://github.com/SoftFever/OrcaSlicer/assets/71825177/41dcf065-1afc-469d-b599-8da5543fb679

Pictures of the rsult:
[image]
[image]
[image]
[image]

### Expected results

The nozzle should not lift and start from wrong location the seam.

### Project file & Debug log uploads

Project file
Log files (No crashes)
Gcode file

[BugReport.zip](https://github.com/SoftFever/OrcaSlicer/files/14628031/BugReport.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

The print behaviour is also visible in the gcode animation.

erdo100, 2024-03-17:
Found my Mistake: I set 20 instead of 20%.
How Can I delete the Bug reoprt?

erdo100, 2024-03-17:
Found my Mistake: I set 20 instead of 20%.

closed 2024-03-17 by erdo100

vgdh, 2024-03-22:
[erdo100](https://github.com/erdo100) I have the best results with the seam gap equal zero.