OrcaSlicer/OrcaSlicer #4534: "Detect narrow internal solid infill" creates random artifacts in corners

adripo, 2024-03-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

W11

### Printer

BambuLab P1P

### How to reproduce

I was trying to print https://www.thingiverse.com/thing:4320201 and came across a strange issue.
The wallet is positioned upside down, with the opening on the bottom.
If the option "Detect narrow internal solid infill" is enabled (enabled by default for me) the slicer will produce strange behaviour on layers 2, 4.
In the corner of the internal "flap" it will create a separate triangle with concentric pattern and will do the rest of the layer with the correct pattern.
This is an issue in my case because the extruder will travel from that triangle to the other side of the layer to continue the infill without retracting and oozing filament that will ruin the print.

[image]
[image]
[image]

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

Attached projects:

[Wallet_issue.zip](https://github.com/SoftFever/OrcaSlicer/files/14628458/Wallet_issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

vgdh, 2024-03-17:
You can disable this feature. Just uncheck **detect narrow internal solid** infill in **strength**

adripo, 2024-03-19:
Why does the behavior differ between Orca and BS despite identical settings? If this is indeed the case, then there appears to be an issue with the print order: after printing the triangle, the extruder travels to the opposite side of the layer to begin the standard monotonic infill. A more logical approach would be to initiate the infill at the triangle and continue towards the other side, or conversely, start on the opposite side and conclude with the triangle.

adripo, 2024-06-19:
@igiannakas did you have time to check?

@SoftFever I just rechecked with version 2.1.0-beta and the results are the same. Please check the attached video. As you see the triangle is done at the end of the layer by moving the head back and forth and ruining the print.
[image]

igiannakas, 2024-06-19:
This is intentional. It’s because that triangle is faster to print than a small zig zag line and also causes less over extrusion compared to a small line.

Bambu studio doesn’t have this feature (along others which have been ported from prusa to orca) so doesn’t do this. But that doesn’t mean that is the better approach ;)

in any case if you don’t want those small triangles you can disable this feature as said above but it’s not generally recommended to do so.

igiannakas, 2024-06-19:
So I’ve found the reason for this - the code basically always adds the narrow concentric infill to the last position in the layer infill - it may make some sense to check the distance from the next solid infill and order them based on proximity. 

I may work on this in the near future but no promises :)

adripo, 2024-06-21:
Thank you for the update. Yes, this would definitely be helpful.

igiannakas, 2024-06-21:
I tried implementing the above and it didn’t fix it - the distance is not mother problem but rather proximity to the last/first position. I’ll try again with a different fix later :)

closed 2024-09-28 by github-actions[bot]

adripo, 2024-10-10:
This issue is not solved yet