OrcaSlicer/OrcaSlicer #4535: X1C LCD object skipping no longer functional

WimVDK, 2024-03-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 beta

### Operating System (OS)

Windows

### OS Version

Win 11 23H2

### Printer

X1C

### How to reproduce

1. Load plate 4 of the attached project file
2. slice
3. send to printer (i never pick send & print)
4. Start print
5. click object skip button on LCD
6. Tap any of the 20 models

### Actual results

ALL 20 objects are being selected at once, not JUST the object being tapped on

### Expected results

Only the object tapped should be selected for skipping

### Project file & Debug log uploads

[GridfinityBox20x20.zip](https://github.com/SoftFever/OrcaSlicer/files/14628563/GridfinityBox20x20.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

larrys, 2024-04-20:
This also happens with my A1 mini.

[This reddit post](https://www.reddit.com/r/BambuLab/comments/1c1d7gf/skip_object_feature_selects_all_objects/) also seems to be the same issue.

ImSadPanda, 2024-04-29:
This has also happened on for my three P1S printers and my one P1P. If I select one image they all light up red or, in the case of a multicolor print, it will isolate just one color across all clones.
You can try and use the tic boxes at the bottom to select the single item that way but it does not show you in the image what item you are about to skip. Its more of a clicking in the dark and "hope you selected the right one" kind of thing.

Orca, a little help here!

scott-c-jensen, 2024-07-30:
I have the same issue. P1S with Orca 2.1.1, multiple identical objects with multicolor.

closed 2025-02-05 by github-actions[bot]