OrcaSlicer/OrcaSlicer #4536: The slicing of this file hangs at 80%

Artemonim, 2024-03-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11 23H2

### Additional system information

[DxDiag.txt](https://github.com/SoftFever/OrcaSlicer/files/14628626/DxDiag.txt)

### Printer

Ender 3

### How to reproduce

The slicing of this file hangs at 80%. If you switch layer from 0.2 to 0.3 - slicing happens as it should.
This is most likely due to Arachna's non-standard settings.

### Actual results

Slicing hangs at 80%. When you try to stop slicing, the slicer freezes.

### Expected results

The slicing is going well.

### Project file & Debug log uploads

[Catapult_CPLT-C1_repaired.zip](https://github.com/SoftFever/OrcaSlicer/files/14628622/Catapult_CPLT-C1_repaired.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

mlewis-everley, 2024-03-18:
I am seeing this issue too. I can't share the file as it is of STL's I purchased, but if I "fill bed with copies" on a smaller item with some complex geometry then try and slice it, I get 80% through and Orca force quits (on Linux).

Hlatbern90, 2024-03-26:
> I am seeing this issue too. I can't share the file as it is of STL's I purchased, but if I "fill bed with copies" on a smaller item with some complex geometry then try and slice it, I get 80% through and Orca force quits (on Linux).
> 
> I am getting this same issue on:
> 
> v1.8.* v1.9.* v2.0.*
> 
> I also tried switching from "Arachne" to "Classic", but Orca also crashed.

I'm seeing the same on my machine (Linux Mint 20.3), but only for v2.0.*.

The complexity of the part does not seem to matter, nor does classic vs Arachne. It gets to 80%, hangs, and force quits.

xMxMiller, 2024-03-29:
I am having a similer issue,

Had it on 1.8, 1.9 and now 2.0.
But it can also crash at 15%.

Could sometimes get around it by making most of the models not printable then one by one making them printable till I have the full plate of models sliced.

anselor, 2024-04-26:
I'm also having a similar issue. Here's the project file:
[Fallout Vault Boy Bobble Head.zip](https://github.com/SoftFever/OrcaSlicer/files/15125446/Fallout.Vault.Boy.Bobble.Head.zip)
Plate 1 freezes at 80%.
If I change layer height to 0.2mm it slices fine.

closed 2024-08-02 by github-actions[bot]