OrcaSlicer/OrcaSlicer #4543: Cut tool button missing from UI in v 2.0.0, keyboard shortcut (Ctrl +C) still works

Ash2033, 2024-03-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

BambuLab P1S

### How to reproduce

In the toolbar buttons above the render window, it seems the cut button was replaced by a Boolean mesh button(which also has not activated/ darkened from grey when selecting two objects), and the cut tool button is no longer present. The keyboard shortcut of Ctrl + C still activates the cut tool on the selected part/object.

### Actual results

[image]

[image]

[image]

### Expected results

I assume although not perfect the cut tool was intended to stay in, and have it's own button next to the boolean mesh tool.

### Project file & Debug log uploads

[debug_Sun_Mar_17_20_52_31_3504.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14629625/debug_Sun_Mar_17_20_52_31_3504.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

It would be cool to have a button again. I will definitely forget the keyboard shortcut in a week and be frustrated without a visual button.

Ash2033, 2024-03-29:
I've found the tool button appears when advanced mode is toggled - That may be intended and I'll mark this closed. Otherwise I'm willing to try to add the button in basic mode if someone can direct me to where those buttons appear in the codebase; I'm not a programmer and not versed in it but I'm willing to try/learn.

valeyard, 2024-05-29:
> I've found the tool button appears when advanced mode is toggled - That may be intended and I'll mark this closed. Otherwise I'm willing to try to add the button in basic mode if someone can direct [...]

Thanks, I had no idea that toggling advanced mode on/off changed the buttons visible here!

tluv2006, 2024-07-31:
> I've found the tool button appears when advanced mode is toggled - That may be intended and I'll mark this closed. Otherwise I'm willing to try to add the button in basic mode if someone can direct [...]

Can we have Advanced  defaulted on please? Thanks for finding this.

emeyedeejay, 2024-08-12:
OMG - thanks for this post ... this was driving me nuts - a friend had not cut icon and we had no clue why. Is cutting an advanced operation? It's not imo - it's a basic thing that many people use to slice things up to fit on build plates.

closed 2024-11-18 by github-actions[bot]

maplemedley, 2025-08-08:
There is no reason to hide this button from every user that doesn't need all the parameters the advanced mode fills the settings pages with Slicing is a bog-standard operation that lets people test-print parts of their model. For reference, BambuLab decided to not hide this button from basic mode. 

People who don't use advanced mode either have to keep advanced mode on, making every settings page five times as long and cluttered with options they're unlikely to use, or they'll have to constantly switch advanced mode on and off, just to access this basic prototyping functionality. 

This is a needless frustration point, its not like there's not enough space on screen for this one button (and if that is the case, plenty other buttons are more advanced than slicing, like brim ears and variable layer height)