OrcaSlicer/OrcaSlicer #4544: Add support for "bambustudioopen://" to Orca Slicer

T3chC0wb0y, 2024-03-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It's frustrating when you go to MakerWorld and cannot download the raw files for a model. In this scenario, you have to click the "open in Bambu Studio" link, save the files as a project or export them as .stl, and then open them in Orca Slicer. This requires you to have Bambu Studio and Orca Slice installed on your machine.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Orca slicer should recognize and support the "open in Bambu Studio" launch command when there is no Bambu Studio instance installed on the machine.

### Describe alternatives you've considered

1. Add an "Open in Orca Slicer" option to Maker World (not likely to happen)

### Additional context

When you click the "open in Bammbu Studio" link from MakerWorld, it make the following call to Open Bambu Studio (if installed): "bambustudioopen://".  Please add feature to Orca Slicer to check for a Bambu Studio installation. "if installed=False then open Orca Slicer else ignore" when the "bambustudioopen://" call is made.

Maximus-CZ, 2024-03-18:
>"if installed=False then open Orca Slicer else ignore" when the "bambustudioopen://" call is made.

I have installed both and still want to open in Orca Slicer by default. There should be a button in settings to associate Orca Slicer with bambustudioopen:// links

anselor, 2024-03-19:
You can modify this yourself with the registry editor.

Save this in a `.reg` file and import it:
```
[HKEY_CLASSES_ROOT\bambustudio\shell\open\command]
@="\"C:\\Program Files\\OrcaSlicer\\orca-slicer.exe\"  \"%1\""
```

Maximus-CZ, 2024-03-19:
Thats awesome to hear. Regardless, I think this still should get included
in Orca so everyone has access to it.

LordBoos, 2024-03-21:
@anselor The registry edit doesn't work for me. Yes, it does open the Orca Slicer, but Orca Slicer fails to download the model and open it.

anselor, 2024-03-22:
@LordBoos 
Sorry, I'd assumed that Orca wouldn't have changed that code from upstream but it seems it has. Digging around I see that the code to handle URLs exists but the installer appears to not add the registry entry. I manually added an orcaslicer registry entry and was able to confirm models open so the code path functions as expected.

The format is supposed to be like this:
`orcaslicer://open?file=[URL GOES HERE]`
So, for example:
`orcaslicer://open?file=https://files.printables.com/media/prints/212195/stls/1944457_811266ef-51c9-4d6d-8f6a-7822ca95d261/sample.stl`

I'd propose amending the original request to optionally handle multiple common slicer url prefixes such as:
* `makerworld://`
* `prusaslicer://`
* `orcaslicer://`

LordBoos, 2024-03-22:
@anselor I tested registry edit (basically copy of bambustudio entry changed to orcaslicer entry and path to orca slicer exe) to add orcaslicer://open which partially works. Then I wrote browser userscript to change `bambustudio://open` URI to `orcaslicer://open` URI when clicking the `Send to Bambu Studio` button on MakerWorld:
```
(function() {
    'use strict';

    document.addEventListener('click', function(event) {
        if (event.target.href && event.target.href.includes('bambustudio://open')) {
            var newUrl = event.target.href.replace('bambustudio://open?file=', 'orcaslicer://open?file=');
            console.log('Clicked on:', event.target.href);
            console.log('New URL:', newUrl);
            event.target.href = newUrl;
        }
    });
})();
```
This indeed opens Orca Slicer and it tries to download the 3mf file, but fails for some reason which I was not able to fix yet.
Orca tries to download the model:
```
:Slic3r::GUI::GUI_App::post_inithttps://makerworld.bblmw.com/makerworld/model/US6b22f37c2643f/55783278/instance/model.3mf?at=1711028136&exp=1711028436&key=8abcac4295312dc068d7efb6a6db4efc&uid=1577422717&name=BenchViseBuildPlate.3mf
```
Then fails:
```
:Error getting: ``: HTTP 0, curl:Couldn't resolve host name:
Could not resolve host: https%3A%2F%2Fmakerworld.bblmw.com%2Fmakerworld%2Fmodel%2FUS6b22f37c2643f%2F55783278%2Finstance%2Fmodel.3mf%3Fat%3D1711028136%26exp%3D1711028436%26key%3D8abcac4295312dc068d7efb6a6db4efc%26uid%3D1577422717%26name%3DBenchViseBuildPla
[Error 6]
```
But if I open the URL in web browser, it downloads the model without issues. I think it might need some authentication or something.

anselor, 2024-03-22:
@LordBoos I tried opening that specific URL and I get a 401 error in my browser. I see there's a uid field in there so I suspect you're right and there's some sort of authentication check to prevent anonymous downloads.

anselor, 2024-03-22:
@LordBoos 
I don't really know javascript so I don't know how to test your snippet but I bet if you change it to remap `prusaslicer://` over you can directly open the links on printables.com just fine. It doesn't require you to log in to download models.

LordBoos, 2024-03-22:
@anselor this link is from my test yesterday and it fails for me too now, it looks like it's also time limited. Looks like timestamp is in `at` parameter.
Try this freshly generated: 
`https://makerworld.bblmw.com/makerworld/model/US6b22f37c2643f/55783278/instance/model.3mf?at=1711119246&exp=1711119546&key=8a5c6a43a335f986163b3f4357a44abc&uid=1577422717&name=BenchViseBuildPlate.3mf`

anselor, 2024-03-22:
@LordBoos 
With the orcaslicer registry entry in place, opening this URL in the browser correctly downloads and opens that 3mf file:

orcaslicer://open?file=https://makerworld.bblmw.com/makerworld/model/US6b22f37c2643f/55783278/instance/model.3mf?at=1711119246&exp=1711119546&key=8a5c6a43a335f986163b3f4357a44abc&uid=1577422717&name=BenchViseBuildPlate.3mf

anselor, 2024-03-22:
Really, thinking about this more, it would probably make more sense for there to be a browser plugin that just lets you remap any slicer url to any specific slicer you have installed rather than depend slicers to support interoperability with other slicer URLs.

LordBoos, 2024-03-22:
This is weird, my Orca Slicer fails to open that link you posted.
[image]
Do you have the same registry entry as I have? 
[image]

anselor, 2024-03-22:
Make sure you also have this higher up
[image]

Other than language differences, pretty sure it's the same at that registry location:
[image]

LordBoos, 2024-03-22:
Yea I have that. And which version of Orca Slicer are you using?

anselor, 2024-03-22:
I have 1.9.1 installed at that location.
I only run the 2.0.0 beta as a stand-alone package since there are some issues preventing me from managing my Bambu printer correctly.

LordBoos, 2024-03-22:
I have only the 2.0.0 beta, can you try it with the 2.0.0? Maybe it's related to that version.

anselor, 2024-03-22:
I can confirm that this is now broken in 2.0.0
[image]

LordBoos, 2024-03-22:
Ok that's "good". At least I know that the message means "it should work in 1.9.1" now and the userscript is working correctly. I will try to create one for printables now.

T3chC0wb0y, 2024-03-25:
> It seems this was previously requested in #3753<https://github.com/SoftFever/OrcaSlicer/issues/3753> - though worded differently.

I saw that prior to adding my enhancement request, but that one seemed to be specifically dealing with MAC and a third-party app used to make changes to allow the launch from MakerWorld.  I opened this enhancement to allow it to be resolved for everyone and eliminate the need for a workaround.

closed 2024-07-01 by github-actions[bot]

anselor, 2024-07-08:
@Rulevoid It should be in 2.1.

https://github.com/SoftFever/OrcaSlicer/pull/5486

https://github.com/SoftFever/OrcaSlicer/pull/5416

https://github.com/SoftFever/OrcaSlicer/pull/3823

https://github.com/SoftFever/OrcaSlicer/releases

Rulevoid, 2024-07-08:
@anselor Confirmed. Looks good in 2.1.1. Thanks!