OrcaSlicer/OrcaSlicer #4545: Gray box for the max acceleration for travel of printer motion ability setting. The box is gray and not editable.

FlashforgeOfficial, 2024-03-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 beta

### Operating System (OS)

Windows

### OS Version

Edition	Windows 10 Pro Version	22H2 Installed on	‎9/‎5/‎2023 OS build	19045.4046 Experience	Windows Feature Experience Pack 1000.19053.1000.0

### Additional system information

Device name	DESKTOP-S17PHJ6
Processor	13th Gen Intel(R) Core(TM) i7-13700F   2.10 GHz
Installed RAM	32.0 GB (31.8 GB usable)
Device ID	C3C59957-656A-4CD7-89AB-337205BA64C3
Product ID	00331-10000-00001-AA589
System type	64-bit operating system, x64-based processor
Pen and touch	Pen support

### Printer

Flashforge AD5M Pro

### How to reproduce

Just run the OrcaSlicer, and click the edit button for printer preset. I can't find a way to make the [max acceleration for travel] box editable. As shown below:
[image]

### Actual results

I am not able to edit and have a bit higher max acceleration for travel for the printer.

### Expected results

The box should be black and editable as the other motion ability settings, such as max acceleration for retracting and extruding etc.

### Project file & Debug log uploads

[Report issue_ max acceleration for travel not editable.zip](https://github.com/SoftFever/OrcaSlicer/files/14632807/Report.issue_.max.acceleration.for.travel.not.editable.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14632811/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]
This is a screenshot for your reference. Thank you!

Luke56784, 2024-04-23:
Same problem here with a Ratrig V-Minion.
I'm limited to 9000 maximum travel speed. Gray, not editable box.

I'm using the Orcaslicer 2.0.0

iz3man, 2024-05-19:
Same here. Greyed out with 20.000mm/s2 which is way too much for my little Ender2 ;)

Luke56784, 2024-05-20:
You can set the value manually in your printers .json file.

add this:

```
    "machine_max_acceleration_travel": [
        "15000",
        "9000"
    ],
```

first value is active, second looks like it's the default one.

matiasgql, 2024-08-16:
It is intentional, [here](https://github.com/SoftFever/OrcaSlicer/issues/5188#issuecomment-2081584890) it explains why it is disabled

closed 2024-11-22 by github-actions[bot]

WindAssassin25, 2026-11-08:
SOLVED! - Change firmware flavour to Marlin 2, Max Accel for Travel becomes editable - Set it to what you want - Change FW flavour back to Klipper, box is greyed out but the stupid warning goes away.

According to Mainsail, Klipper is now adjusting to the higher accel I've set in the speed section of Orca Slicer when travelling.