OrcaSlicer/OrcaSlicer #4547: Issue with importing filament & printer profiles after laptop change

fryc88, 2024-03-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Win 11

### Additional system information

Not relevant.

### Printer

Voron 2.4, Creality K1 Max, Sovol SV01 pro

### How to reproduce

1. File > Export configs.
2. Reinstall windows/change PC
3. File > Import configs

### Actual results

Nothing gets imported.
[image]

### Expected results

All Orca, printer and filaments configs to be imported.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14633652/log.zip)
[Klicky-dock-A-no_logo.zip](https://github.com/SoftFever/OrcaSlicer/files/14633653/Klicky-dock-A-no_logo.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Export/import config seems to be completely broken.

cdreier, 2024-03-30:
did you manage to import your configs somehow? just hitting the same problem on linux :(

closed 2024-07-06 by github-actions[bot]

jrbenito, 2025-02-17:
Same here is happening with Creality K1, Linux Appimage Orca

pm0u, 2025-03-12:
Yep. Export a printer, import it on another machine... nothing? Is there some trick here? I was hoping that the printer preset would be the printer and everything that goes along with it

pm0u, 2025-03-12:
I managed to import mine by adding the generic system preset that relates to my printer and then import the saved presets. This did include all filaments and print profiles thankfully. I'm not sure what you would do for a generic - i guess start with the generic klipper or marlin profile?

dyaln-hilger, 2025-09-06:
I know this thread is pretty old, but I ran into the same issue just moments ago when moving files & configs over from my old desktop. I found that you need to import your .orca_printer config files to the SAME version of OrcaSlicer that they were taken from.

For example, my old desktop had OrcaSlicer v2.0.0; when setting up my new desktop I just installed the latest version from OrcaSlicer's website (v2.3.0), this resulted in the .orca_printer config files not being properly import/register within the application. I tried importing to an older version (v1.9.0), but this also didn't work.

It was only when I completely uninstalled the previous versions of OrcaSlicer, restarted my PC, and then installed the same OrcaSlicer version from my old desktop (v2.0.0) that the .orca_printer config files were completely and properly imported and available within the app. 

There was no warning or indication that a version mismatch was the exact cause, but through trial and error I'm fairly confident that was the cause. 

I haven't yet tried updating, but I'm sure now that they exist, I'll have a better chance of going to a newer version.
Hopefully this helps. Best of luck!

Silezmogul, 2025-11-30:
Still present, although I managed to get export/import working.

If you want to export your old printer config to the new program/device, first go to Prepare tab > Printer > Edit printer preset > Connect to your device.

This is when all my printer, filament and process presets appeared. In your case, might want to try to import after that.