OrcaSlicer/OrcaSlicer #4553: Slow Down For Overhangs - Artifacts induced by unwanted slowdowns.

lhndo, 2024-03-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

LH Stinger

### How to reproduce

Load the file
Switch preview to Flow and observe unwanted slowdowns continuing out of the overhang areas.
Switch to Classic and observe they are gone.

### Actual results

[image]

### Expected results

No slowdowns on perimeters with no overhang
More like Classic mode
[image]

### Project file & Debug log uploads

[cube overhang.zip](https://github.com/SoftFever/OrcaSlicer/files/14638331/cube.overhang.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lhndo, 2024-03-28:
Hey @SoftFever, just curious, what does make this an enhancement as opposed to a bug?

SoftFever, 2024-03-28:
These are single ‘long’ extrusions (G1 command) which have only a small part overhanging. Therefore, the whole extrusion is considered as overhanging. To properly apply overhang settings to the exact part of the extrusion, we will need to do some extra work first to break this one ‘G1’ command into multiple smaller G1 extensions.And this might have other consequences like generating larger gcode Sent from my iPhone

lhndo, 2024-03-28:
Understood and thanks for taking the time to answer! I will be trying to stay with the classic algorithm, or find a workaround in the mean time.

closed 2024-08-01 by SoftFever