OrcaSlicer/OrcaSlicer #4559: Make Overhangs printable and printing Multi color adds color all the way down

Aghost917, 2024-03-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-Beta

### Operating System (OS)

Windows

### OS Version

Windows 11 22H2

### Printer

Bambu Labs P1S

### How to reproduce

Import the STL file, use the paint tool on the STL, and turn on Make Overhangs Printable.

### Actual results

For some reason, it just adds color all the way to the build plate instead of starting at the layer height of the color change.
[image]

### Expected results

[image]

### Project file & Debug log uploads

[Rubber_Duck.zip](https://github.com/SoftFever/OrcaSlicer/files/14641051/Rubber_Duck.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

okirisx2, 2024-04-04:
Yes. This has been a long-standing bug. I’ve not found a workaround.

closed 2024-07-11 by github-actions[bot]

tomzdotorg, 2024-08-14:
`this->apply_conical_overhang();`

has to be above 

`apply_mm_segmentation(...`

in PrintObjectSlice.cpp

i.e. 

```
    this->apply_conical_overhang();

    // Is any ModelVolume MMU painted?
    if (const auto& volumes = this->model_object()->volumes;
        m_print->config().filament_diameter.size() > 1 && // BBS
        std::find_if(volumes.begin(), volumes.end(), [](const ModelVolume* v) { return !v->mmu_segmentation_facets.empty(); }) != volumes.end()) {

        // If XY Size compensation is also enabled, notify the user that XY Size compensation
        // would not be used because the object is multi-material painted.
        if (m_config.xy_hole_compensation.value != 0.f || m_config.xy_contour_compensation.value != 0.f) {
            this->active_step_add_warning(
                PrintStateBase::WarningLevel::CRITICAL,
                L("An object's XY size compensation will not be used because it is also color-painted.\nXY Size "
                  "compensation can not be combined with color-painting."));
            BOOST_LOG_TRIVIAL(info) << "xy compensation will not work for object " << this->model_object()->name << " for multi filament.";
        }

        BOOST_LOG_TRIVIAL(debug) << "Slicing volumes - MMU segmentation";
        apply_mm_segmentation(*this, [print]() { print->throw_if_canceled(); });
    }

```

GarlicDebug, 2024-09-24:
I just attempted the fix mentioned by @tomzdotorg and it didn't work for the model I tested with.

JamieWritesCode, 2024-10-03:
@SoftFever Can we get this re-opened and looked at again? This is still an issue as of today.

tomzdotorg, 2024-10-03:
> This is still an issue as of today.

I believe the fix I posted above was merged in #6896

JamieWritesCode, 2024-10-03:
Awesome, great stuff! Apologies for the unnecessary tag. I must have missed that PR when searching.