OrcaSlicer/OrcaSlicer #4563: Only the first extrusion length for small area infill compensation model can be 0.

3D-Frank, 2024-03-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.0.0-beta

### Operating System (OS)

macOS

### OS Version

13.6.4 (22G513)

### Additional system information

iMac, Retina 5K, 27-inch, 2017
3,5 GHz Quad-Core Intel Core i5
Radeon Pro 575 4 GB
8 GB 2400 MHz DDR4

### Printer

Bambu P1S

### How to reproduce

I always get the following error message when I want to slice any model:
"Only the first extrusion length for small area infill compensation model can be 0"
No matter which printer, filament or process setting I choose
[image]

### Actual results

XY

### Expected results

XY

### Project file & Debug log uploads

Xy

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

XY

eforulus, 2024-03-20:
I also faced this error, but when I changed the language from Ukrainian to English and solved this problem,

version 2.0.0 - beta

3D-Frank, 2024-03-20:
Thank you very much for the tip! I'll test it and report back afterwards

3D-Frank, 2024-03-20:
When I changed the language from German to English the error disappears

Visnitz, 2024-03-22:
In M1 Air, OS X 13.6.5 solve this problem change Hungarian to Englis.

AstafievSO, 2024-03-30:
> попробуйте новую версию release candidate, возможно, она была решена [OrcaSlicer_Windows_V2.0.0-rc_portable.zip ](https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds)
no, the error still persists, I tried
OrcaSlicer_Mac_x86_64_V2.0.0-beta.dmg
OrcaSlicer_Mac_x86_64_V2.0.0-rc.dmg
OrcaSlicer_Mac_x86_64_V2.0.0.dmg

Basti10785, 2024-04-08:
[image]
Unfortunately, after updating to version 2.0.0, I also have the error message. The current language is set to German, but when I switch to English the error message doesn't appear. But I would still like to use it in German, so a quick fix would be desirable.

iStarComLab, 2024-04-08:
https://github.com/SoftFever/OrcaSlicer/issues/4891
https://github.com/SoftFever/OrcaSlicer/issues/4480

The problem is old and there is still no solution!

Basti10785, 2024-04-29:
Hi,
Is there still no practical solution to this problem? Because I would like to use the program in German and not in English just because something doesn't work correctly.

iStarComLab, 2024-04-29:
> Hi,
> Is there still no practical solution to this problem? Because I would like to use the program in German and not in English just because something doesn't work correctly.

Es gibt eine Lösung und sie ist unten! Löschen Sie einfach alle Werte im Eingabefenster und speichern Sie Ihre Voreinstellung! Genießen Sie es für Ihre Gesundheit!
[image]

iStarComLab, 2024-04-29:
[image]

CarloJenna, 2024-05-09:
I have the same problem with the Italian idiom and I can't find the tab that you say.

> > Hi,
> > Is there still no practical solution to this problem? Because I would like to use the program in German and not in English just because something doesn't work correctly.
> 
> Es gibt eine Lösung und sie ist unten! Löschen Sie einfach alle Werte im Eingabefenster und speichern Sie Ihre Voreinstellung! Genießen Sie es für Ihre Gesundheit! ![Снимок экрана 2024-04-29 в 15 12 46](https://private-user-images.githubusercontent.com/129138485/326402472-0a89af98-fe56-4278-9d71-fb6f2058ca57.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyNjE4MTQsIm5iZiI6MTcxNTI2MTUxNCwicGF0aCI6Ii8xMjkxMzg0ODUvMzI2NDAyNDcyLTBhODlhZjk4LWZlNTYtNDI3OC05ZDcxLWZiNmYyMDU4Y2E1Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwOVQxMzMxNTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zNWVlZTdiYjQ4MjY0ZGM2OTA1YzgzMmFkNDZjNGFhYmZkZjg5ZWY0ZTgwOTJhYTc0OWYxNTI5ZTNmY2E4MWY2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.D-phmqoViZGsMukofIxZKpWCbKGeD8NyS5bgHkp5mdU)

adiazsanti, 2024-05-09:
returning from spanish to english worked

CarloJenna, 2024-05-10:
> returning from spanish to english worked
Yes I did it!! ahahahahahah but I want to stay with my Italian ahahahah (or Catalan)

Basti10785, 2024-05-10:
[image]
Activate "small area flow compensation" and delete the values from the window and save the profile again.

CarloJenna, 2024-05-10:
done it!! thank you very much

closed 2024-08-17 by github-actions[bot]