OrcaSlicer/OrcaSlicer #4564: Crash when choosing side view

tintad, 2024-03-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Qidi X-CF Pro

### How to reproduce

1. Open slicer
2. Choose side view (Front, Rear, Left, Right view) or Short cut Ctr+3, ...
3. Zoom in and zoom back out

### Actual results

[image]

### Expected results

Model will disappear, have to restart the Slicer for it to work again

### Project file & Debug log uploads

[Error.zip](https://github.com/SoftFever/OrcaSlicer/files/14641489/Error.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

mdeeter, 2024-04-03:
I am also seeing this on Mac using version 2.0.0.  Super annoying.  

I'll use the shortcuts (like CTRL-0) to change the view... after the screen updates the view, if I try to zoom or move the view, then the model disappears and I have to restart the app.

mdeeter, 2024-07-03:
.... bad bot... the problem still exists

closed 2024-10-09 by github-actions[bot]