OrcaSlicer/OrcaSlicer #4570: Rename custom filament profile

CryPt00n, 2024-03-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Renaming custom profiles is currently not possible in an easy way, or in total I think. when saving a profile with a different name, there is a second, inherited profile created. So this will just flood the profiles.
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

A function to rename custom filament profiles.

### Additional context

If a solution or workaround is available for this, I would appreciate it.

gshpychka, 2024-03-28:
Duplicate of https://github.com/SoftFever/OrcaSlicer/issues/3669, which was already a feature request for this

CryPt00n, 2024-03-29:
> Duplicate of #3669, which was already a feature request for this

As I mentioned in the description, I´m aware of this case and have also linked it. But it´s not tagged correctly, so I created a feature request.

gshpychka, 2024-03-29:
> > Duplicate of #3669, which was already a feature request for this
> 
> As I mentioned in the description, I´m aware of this case and have also linked it. But it´s not tagged correctly, so I created a feature request.

the solution is to tag that issue, not create another one

sk8board, 2024-06-18:
Not being able to rename a user preset is a basic function that a user would expect. Why would this be tagged as an enhancement rather than a bug?

Brumes-Wolf, 2024-09-19:
This request is still relevant and the whole filament adding process is rather user unfriendly and pretty infuriating.

MillyRamsey, 2024-11-26:
This is something that is very frustrating for me, too. I want to have my filament list nice and neat, and as it is (creating new filaments all the time with pre-generated names, then trying to rename them), it's a mess!

closed 2025-03-05 by github-actions[bot]