OrcaSlicer/OrcaSlicer #4571: Cannot print to K1C printer

phuongnd08, 2024-03-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

macOS

### OS Version

14.3.1 (23D60)

### Printer

Creality K1C

### How to reproduce

0. Add K1C Printer, update IP Address
1. Check that the printer successfully connected in the Device tab
[image]
3. Prepare a model
4. Slice 
5. Print > Upload

### Actual results

[image]

```
Error uploading to print host:
HTTP 404: <!DOCTYPE html>
<html>
<head>
<title>404 Not Found</title>
</head>
<body>
<center><h1>404 Not Found</h1></center>
<hr>
</body>
</html>
```

### Expected results

G-Code file uploaded successfully

### Project file & Debug log uploads

Project File:
[NutTapLuoiError.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14643900/NutTapLuoiError.3mf.zip)

Log File:
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14643919/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bistory, 2024-03-19:
You need to root your printer first

phuongnd08, 2024-03-23:
K1C printer is rooted by default. I just checked, the printer is already rooted, at least on the interface. I can get the root password going to System > Root account information.

bistory, 2024-03-23:
> K1C printer is rooted by default. I just checked, the printer is already rooted, at least on the interface. I can get the root password going to System > Root account information.

Update moonraker, that's what you need to be able to use push gcode from orca.

phuongnd08, 2024-03-23:
I installed moonraker:

[image]

But uploading g-code still fails:

[image]

phuongnd08, 2024-03-23:
I still see Creality default web interface so I'm not sure if there is something else to be done:

[image]

phuongnd08, 2024-03-23:
Okay, I updated the host to http://xxx.xxx.xxx.xxx:4408/ and it's uploading just now. Many thanks @bistory for your support

closed 2024-03-23 by phuongnd08

paulochavesBrasil, 2024-10-15:
> Okay, I updated the host to http://xxx.xxx.xxx.xxx:4408/ and it's uploading just now. Many thanks @bistory for your support

how do you do it? same problem here, can t send g code direct from my k1

phuongnd08, 2024-10-17:
[image]
See these red dude.