OrcaSlicer/OrcaSlicer #4575: 1 layer thickness on top / bottom

ValikDov, 2024-03-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

e.g. 2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Anycubic Kobra 2 

### How to reproduce

No Idea how to reproduce

### Actual results

On the screen you can see layer settings that should be used 
[image]
1 top layer height
[image]
Still 1 layer height 
[image]
[image]

bottom

### Expected results

Heigth level of 5 layers on top and bottom surfaces

### Project file & Debug log uploads

[Layer_bug_2.0.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14648901/Layer_bug_2.0.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ValikDov, 2024-03-19:
There will be a question about "Only one wall of top surfaces". This tab is disabled

Sophist-UK, 2024-03-20:
I too am seeing this issue - I ask for multiple top and / or bottom layers, but the slicer only slices a single top or bottom layer.

kayboy-bebop, 2024-03-21:
I can reproduce this error when "Ensure vertical shell thickness" is set to "None".

lhndo, 2024-03-21:
same here as @kayboy-bebop . "None" setting and 0 shell thickness creates a single top/bottom layer.

ilko-k, 2024-03-23:
I was about to report this and a quick search took me here so I can confirm. When I want 0 infill, all the shell's solid fill is not generated if "Ensure vertical shell thickness" is disabled ( Strenght → Advanced ).

vgdh, 2024-03-23:
I can't reproduce the issue in release candidate version

kayboy-bebop, 2024-03-23:
> I can't reproduce the issue in release candidate version https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds try latest RC version

Yup, RC release fixed this for me.

vgdh, 2024-03-23:
[ValikDov](https://github.com/ValikDov) please check and close the issue if it resolved.

ilko-k, 2024-03-24:
latest nightly seems ok in my case (build from 2024.03.23)

closed 2024-06-30 by github-actions[bot]