OrcaSlicer/OrcaSlicer #4578: Can't export config files

missile150, 2024-03-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.0

### Operating System (OS)

Windows

### OS Version

n/a

### Printer

Ender 3 Pro

### How to reproduce

1.  Go to Export Config
2. Select Filament bundle(.orca_filament), or Select Filament presets(.zip), or select Process presets(.zip)
3. You should be able to select printer names for these files.  That function does not exist and will not let you export any of the 3 items in step 2.

### Actual results

Can not export 3 of the 5 export selections.

### Expected results

We should be able to export all Presets.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

[image]

[image]

[image]

Noisyfox, 2024-03-19:
This has been fixed in 2.0

Noisyfox, 2024-04-11:
Could you send me your "C:\Users\%username%\AppData\Roaming\OrcaSlicer" folder?

Noisyfox, 2024-04-14:
> Sure, if it'll help solve this, but how?

Email me via timemanager.rick (at) gmail

h-overstreet, 2024-07-02:
> This has been fixed in 2.0

I have this issue in 2.1.1

dderbidge, 2024-07-10:
> This has been fixed in 2.0

I made a mistake. I somehow typed the wrong version. I'm using 2.1.0 on a Mac.

mbussen-ot, 2024-07-18:
I can also confirm that this issue is still happening on 2.0.0 and 2.1.1.

dderbidge, 2024-07-18:
I should have clarified that my issue was due to this issue.
#6019 

@mbussen-ot Can you post a screenshot of what you are seeing in 2.1.1? Also, are you using a Mac too?

mbussen-ot, 2024-07-18:
I am not using a Mac.  I am on Windows 11 Enterprise 22000.3079.  

As an example, I have User Presets for my Process configuration, but when I attempt to export them, it doesn't give me the option to select any presets on the Export Preset Bundle window.  If I press "OK" it gives me the message that I need to select an option.

[image]
[image]
[image]

5jvm0u4, 2024-08-09:
Same problem here, there's no way to scroll down to hit those hitboxes
[image]

Mosher23, 2024-09-06:
Same here cannot select anything for the export.

Mac OS

jaeseopark, 2024-09-11:
Same issue. I have Bambu A1 and A1 Mini. Only Bambu A1 Mini is available for selection in the export window. I have custom filaments for both printers.

Rickthebig, 2024-09-12:
I don't want to be boring but... same issue here. I use a self-defined printer based on changes from an existing one. I cannot choose anything to export (v. 2.1.1 on Windows).

bradtn, 2024-12-07:
Fix this issue please I can't see my custom printer or filaments I don't get why or how this is an issue still and so many versions later

MASantangelo, 2025-01-30:
I am having this issue in 2.2.0 but I think I found a workaround.

So when I have my printer selected, if I go to export, and the currently selected printer does not show up in the list. 

[image]

However, if I select another printer, it now mysteriously shows up in the list.  And the newly selected one somehow still remains in the list (I thought it would disappear, it did not)

[image]

Try adding and selecting a different printer and then trying to export?

bartek-swiderski92, 2025-02-23:
I am also having this issue, however only for filament and process presets. Printer works just fine, also the above work around doesn't work for me.

erickCantu, 2025-03-13:
@MASantangelo Your "solution" works. I had to add a dummy printer, since I only have one.

GinToncek, 2025-04-13:
Please start working on this issue. This got me delete a printer with filament files, without having an actual copy of filament profiles.

afewyards, 2025-07-20:
I got the same issue here. cannot export user created printer.

5jvm0u4, 2025-07-20:
Could someone reopen this?