OrcaSlicer/OrcaSlicer #4579: Object disappears from canvas when panning, zooming, orbiting etc. from an orthographic camera position.

MyCatMoss, 2024-03-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

14.2.1

### Additional system information

MacBook Pro M3 Max, 48gb RAM

### Printer

X1C

### How to reproduce

1. Open 3d model
2. Press CMD+3 or CMD+4 to view model straight on
3. Attempt to zoom in or out using scroll wheel or even pan

### Actual results

Anytime I try to adjust the view (pan, orbit, zoom in/out) while viewing my model **from an orthographic position** (CMD+1, CMD+2, etc.), it causes the model to completely disappear from the canvas. When it happens, it almost seems like I zoomed in too far. However, no amount of zooming out allows me to find the model again. I am forced to restart the program each time this happens.

### Expected results

You should be able to simply zoom in to see the model closer up.

### Project file & Debug log uploads

[Orca Log.zip](https://github.com/SoftFever/OrcaSlicer/files/14650077/Orca.Log.zip)
[Bike Grips v4-1 - BUG.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14650082/Bike.Grips.v4-1.-.BUG.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Screenshot of canvas after item has disappeared.
[Screenshot.zip](https://github.com/SoftFever/OrcaSlicer/files/14650106/Screenshot.zip)

nirurin, 2024-03-27:
Same issue here. Has the cause been figured out?

foreachthing, 2024-03-27:
Same on Windows 10. O-S 2.0b

_Btw, this is still a "perspective" view, not "orthographic"._

AxelDreher, 2024-03-30:
The problem also exists in the current version of Orca Slicer 2.0.0 (Win Final Version).

Set orthogonal view.
Insert a primitive cube.
Press Alt gr + 4 for the front view.
And zoom in!

Everything disappears

buzzhuzz, 2024-04-01:
Duplicates #3615

ChevaTomme, 2024-06-01:
Happens to me sometimes as well.

The user might accidentally move the objects outside the window as well and a function to re-center (View all, Zoom all...) the objects in the 3D window would be useful and solve the need for re-starting Orca every time this happens.

closed 2024-09-08 by github-actions[bot]