OrcaSlicer/OrcaSlicer #4583: Blowing the non-hanging overhang

Artemonim, 2024-03-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11 23H2

### Printer

Ender 3

### How to reproduce

Slice a cylindrical model, with elements on the support

### Actual results

The fan blows the entire support layer equally, resulting in poor baking of the layer.
Pics with fan heatmap:
[image]
[image]

### Expected results

Ensure airflow is directed at the sections printed on support structures, but avoid airflow on sections being printed directly on top of the previous layer of the object itself. This is especially important for filaments that are harmed by blowing.
[image]

### Project file & Debug log uploads

[gear.zip](https://github.com/SoftFever/OrcaSlicer/files/14652324/gear.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Panzerhalzen, 2024-03-26:
A similar problem. The fan blows the entire layer at full power, although it should only work on the supports. In addition, after printing the internal bridges, the fan does not turn off and continues to work until the first layer of the cover is printed.
[image]
[image]

Artemonim, 2024-04-04:
Haven't re-tested the case from the report, but a similar problem is caused by shifting the timing of the blower fan on.

closed 2024-07-11 by github-actions[bot]