OrcaSlicer/OrcaSlicer #4584: Monotonic concentric top layer pattern.

MakeFancy, 2024-03-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm always frustrated when i have a top layer that prints starting from the outside of the surface then skipping to somewhere in the middle of the surface only to leave a gap in between which would eventually be printed later. 
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

just make it so the top layer concentric paths can print so that every line overlaps an adjacent already printed lines so, strictly outside to inside or inside to outside depending on where the wall paths are. there could also be a spiral pattern the follows the perimeters of the walls instead of forcing a circular shape. 

I've quickly modeled the 2 ideas i have as a visual 
[image]

### Describe alternatives you've considered

I've considered using Archimedean spiral as a top layer pattern but that doesn't work for anything that isn't a perfect circle. 
[image]

DVSVIDEO, 2024-03-19:
I do not understand the problem.
I selected the concentric fill and it works exactly as you want it to, as you modeled your solution...
[image]

MakeFancy, 2024-03-19:
does it print in order? from outside to inside?
and could you tell me why it makes these small triangles on the corners? 
[image]

DVSVIDEO, 2024-03-19:
I can't tell because you didn't attach what you sliced.
In my case it filled in nicely with concentric squares.

MakeFancy, 2024-03-19:
[image]
this is what the object looks like, i am unsure about how to attach the stl or 3mf file. 
I should also mention that i am using this for a large scale printer with a 4mm nozzle diameter

DVSVIDEO, 2024-03-19:
This body has no upper surface. What you set as a top surface in the menu is a flat surface. This one has an inward sloping sidewall.
At a certain angle of inclination, it is considered a top surface and continues from there...

you just drag the 3mf file into the window under the text.

MakeFancy, 2024-03-19:
[image]
this is what it says when i try to attach.

and yes i know it isn't the uppermost surface, i am calling it a top surface because that is what the slicer calls it

DVSVIDEO, 2024-03-19:
for example the same cube, but not horizontally :
[image]

DVSVIDEO, 2024-03-19:
"and yes i know it isn't the uppermost surface, i am calling it a top surface because that is what the slicer calls it"

No. The top surface is not the contour.
In your model the top surface is the tiny last line of the body (the tip)
a pointed body has no upper surface, only a tilted wall.
Try slicing a body with a flat top and you will see that what you have reported as a defect is not a defect.

MakeFancy, 2024-03-19:
[image]
the slicer calls it a top surface because it is a gap between walls on 2 sequential layer

DVSVIDEO, 2024-03-23:
I say there is no problem. The top surface is concentric and regular:
[image]

MakeFancy, 2024-03-25:
> I say there is no problem. The top surface is concentric and regular: ![toplayer](https://private-user-images.githubusercontent.com/96373457/316228227-9731ebba-2213-4913-9ec2-3538a4db025e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEzNzU2NjEsIm5iZiI6MTcxMTM3NTM2MSwicGF0aCI6Ii85NjM3MzQ1Ny8zMTYyMjgyMjctOTczMWViYmEtMjIxMy00OTEzLTllYzItMzUzOGE0ZGIwMjVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI1VDE0MDI0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM3YTc4NzE2ZjM5N2MzNmNiMTZkNjhhMjU4M2EwMjk4MTdkNGNjM2YxZWI4NTZlYTY2MTI5MDc4NWMwOWM4YzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.98ymmY8K9O9AHL2pc5JtnHWcWYlV5dluC1my2Plzn7Y) You just don't understand how the slicer works.

[image]
Here you can see I've unchecked the display for everything EXCEPT "TOP SURFACES" 

[image]
and here, just in case didn't see this the first 5 times friend

DVSVIDEO, 2024-03-30:
Then you have wrongly formulated the cause of the error and therefore the title does not refer to the problem.
The wrong question can only have the wrong answer....
The real problem is that the slicer sinews the top surface but fills it as a sidewall.
And only fills the top surface with the specified pattern.
The real question is where does the top surface of such a body start and how far is it just a sidewall...
How would you decide?

TexZeTech, 2024-04-02:
> <img alt="image" width="775" src="https://private-user-images.githubusercontent.com/162491593/314164836-16eeb478-69f3-44aa-ba04-b9276eda7d60.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIwODI4MDksIm5iZiI6MTcxMjA4MjUwOSwicGF0aCI6Ii8xNjI0OTE1OTMvMzE0MTY0ODM2LTE2ZWViNDc4LTY5ZjMtNDRhYS1iYTA0LWI5Mjc2ZWRhN2Q2MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwMlQxODI4MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZmE3NzlkNTE3MDg1OTZlNTk0NzZlMzllN2U4OTBiY2U4N2UxMGEzM2VlOWYzMTdlZWViMmFmYjUzMmYxYjQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2qpo1wwylxNfK4nFdOHBeHUvYOaUAbF9euS6hSP-gZ0"> this is what it says when i try to attach.
> 
> and yes i know it isn't the uppermost surface, i am calling it a top surface because that is what the slicer calls it

So either put the file into a .zip or just add .zip to the end of the .3dm/.stl file?

closed 2024-09-30 by github-actions[bot]

evnp, 2026-10-27:
+1

The gap doesn't show up in the slicer, just the final print, where the two directions of concentric paths finally meet in the middle.

Seems likely this happens when the shape width isn't an even multiple of layer width (though there are almost certainly other factors during actual print). I'm not sure if 1-way concentric would help but it seems like a great set of options to have. Ideally, four of them:
- middle-in concentric (current algo)
- middle-out concentric (inverse)
- outside-in concentric (start on outside of shape and always work inwards)
- inside-out concentric (start on inside of shape and always work outwards)