OrcaSlicer/OrcaSlicer #4590: Tree support (Slim) not behaving correctly

srbjessen, 2024-03-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Biqu BX Klipper

### How to reproduce

Slicing an object in BambuLab, compared to OrcaSlicer yields very different results.

### Actual results

[image]

### Expected results

Although they are 2 "different" slicers, i was expecting to see more similar results than the actual result.

### Project file & Debug log uploads

[H4-Stole-Donna.zip](https://github.com/SoftFever/OrcaSlicer/files/14657791/H4-Stole-Donna.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tlhintoq, 2024-03-21:
I'll add to this rather than make a new similar bug.
Support selection just isn't working in 2.0 beta.  No matter what you pick you get the typical organic tree support.
Which also means you can't do variable layer height because those two don't go together.
[image]
[image]

tntfriesen1, 2024-03-21:
I am **unable** to replicate this issue on 2.0.0-dev with the attached 3mf file as well as using my own profile and settings.

tlhintoq, 2024-03-24:
[image]
Clearly set to normal|Manual and getting tree supports.
[Best+Poop+System.zip](https://github.com/SoftFever/OrcaSlicer/files/14733948/Best%2BPoop%2BSystem.zip)

closed 2024-06-30 by github-actions[bot]