OrcaSlicer/OrcaSlicer #4591: Request for Flashforge Guider 2s printer profile

AC5SHOT, 2024-03-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I unfortunately can't find a profile for the Flashforge Guider 2s

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Addition of the Flashforge Guider 2s

cochcoder, 2024-03-29:
@AC5SHOT 

Can you upload the profiles for the Guider 2s, through screenshots or copying and pasting? Whenever I try to access the Flashforge slicer, it ends up crashing on me.

AC5SHOT, 2024-03-30:
hey i uploaded it here
on github i don;t know why but i was having issues uploading directly
https://github.com/AC5SHOT/guider-2s/tree/main/slice%20profiles%20and%20a%20benchy%20gcode

cochcoder, 2024-03-30:
> hey i uploaded it here
> 
> on github i don;t know why but i was having issues uploading directly
> 
> https://github.com/AC5SHOT/guider-2s/tree/main/slice%20profiles%20and%20a%20benchy%20gcode

Thanks, I'll look into implementing these when I can! Please make sure to include the printer's max speed/acceleration if you haven't already.

AC5SHOT, 2024-03-30:
awesome thanks so much you genuinely have no idea how grateful i am pls feel free to reach out should anything else be required , i made sure to add the max speed/acceleration on both the Flashforge software cap and the actual machine hardware capabilities as you asked

DirectDesignNZ, 2024-04-15:
I'm also in this boat, thanks all for the work you've put in so far 🙌

cochcoder, 2024-04-18:
@AC5SHOT @DirectDesignNZ What are the G-Codes are used in FlashForge slicer for this printer?

AC5SHOT, 2024-04-26:
@cochcoder @DirectDesignNZ 

im honestly not able to ascertain anything abiout the type of gcodes other than the fact that its labeled as 

gcode flavor :flashforge   and its default output is a x3g output labeled as GX/G File ( i would guess marlin based of the output and production date)

additional nozzle sizes : 0.3, 0.4(default), 0.6, 0.8
the machine can take these types of filament :
ABS, ASA, PA6/66, PA12-CF (9891BK), PAHT (9875NT), PC, PET-CF (9780BK), PETG, PLA, TPU 95A

eveything else i got is here
https://github.com/AC5SHOT/guider-2s/tree/main/slice%20profiles%20and%20a%20benchy%20gcode

sorry for the late reply rough week

cochcoder, 2024-04-26:
> @cochcoder @DirectDesignNZ
>
>
>
> im honestly not able to ascertain anything abiout the type of gcodes other than the fact that its labeled as
>
>
>
> gcode flavor :flashforge   and its default output is a [...]

Don't be sorry, we all have that one week. I'll use the given information and hopefully it can be ready for testing within a week. I did just find a Cura profile online just bow, so hopefully I can borrow the G-Code and have it all work in Orca Slicer.

cochcoder, 2024-05-01:
Can you please try and see if you're able to successfully print with this [build](https://github.com/cochcoder/OrcaSlicer/actions/runs/8912320699)? I doubt that you would be able to as it seems that you need a post-processing script (which I can't add to default profiles), but I figured it's worth a shot.

AC5SHOT, 2024-05-01:
awesome thanks  will try tonight

AC5SHOT, 2024-05-01:
hey cochcoder sorry it didn't work, it keeps saying 
error : relative extruder addressing requires resting the extruder position at each layer to prevent loss of floasting point accuracy. Add "G92 E0" TO LAYER_GCODE.
 
JUMP TO (BEFORE LAYER CHANGE GCODE)

i did try to add this post processing script after it didnt work but this also didnt work sorry

Search	M104 S(.*)\.0
Replace	M104 S\1
Use RegEx	Yes ;
Search	M140 S(.*)\.0
Replace	M140 S\1
Use RegEx	Yes ;

cochcoder, 2024-05-02:
> hey cochcoder sorry it didn't work, it keeps saying
>
> error : relative extruder addressing requires resting the extruder position at each layer to prevent loss of floasting point accuracy. Add "G92 [...]

I think that the post process script needs to be in python for Orca Slicer, the one listed is supposed to work on Cura. Curious though, what happens if you put `G92 E0` in before layer change gcode in the printer's settings?

AC5SHOT, 2024-05-03:
running further tests now will let you know a lil later

AC5SHOT, 2024-05-03:
hey with the G92 E0 fix it keeps outputting "failed to generate g-code for invalid custom g-code

machine start code parsing error at line 2 : not a variable name m140 s{material_bed_temp_layer0}; set initial bed temp

pls check the custom go code or usde the default gcode"   

sorry couldn't find a remedy for that myself

cochcoder, 2024-05-03:
> hey with the G92 E0 fix it keeps outputting "failed to generate g-code for invalid custom g-code
>
>
>
> machine start code parsing error at line 2 : not a variable name m140 s{material_bed_temp_layer0}; [...]

I think I know how to fix that issue, I'll get back to you on that. Also I have one idea for including the post process script if needed, but it'll probably not work.

cochcoder, 2024-05-04:
Here is a new [build](https://github.com/cochcoder/OrcaSlicer/actions/runs/8946266650) that fixes the G-Code errors on my end

AC5SHOT, 2024-05-04:
hey im still issue gona assume its just the repo hasn't pulled yet and will try again tomorrow

cochcoder, 2024-05-04:
> hey im still having the issue gona assume its just the repo hasn't pulled yet and will try again tomorrow

Did you download the new build?

AC5SHOT, 2024-05-04:
yup clean installed on 2 separate laptops just to be safe

cochcoder, 2024-05-04:
Strange...

AC5SHOT, 2024-05-04:
and now the printer is checked but unusable on windows, its only working on macos

AC5SHOT, 2024-05-04:
IDK what happend but its still checked in the printer selection but not available as a printer once it is checked and has now completely not listed on macos at all

AC5SHOT, 2024-05-05:
any chance you could maybe trigger another update ?

AC5SHOT, 2024-05-05:
it just started working gona run a few test prints, nothing went wrong while slicing

AC5SHOT, 2024-05-05:
ran 3 different slices and all of them only print at the top right and of the the build plate

AC5SHOT, 2024-05-05:
for reference its at the home position 

the home position is 

x 156
y125
z300

cochcoder, 2024-05-06:
I've started a new build [here](https://github.com/cochcoder/OrcaSlicer/actions/runs/8970824590) if needed, but you could also enter these values into the slicer to start testing: 
[image]

cochcoder, 2024-05-06:
FYI, if we manage to get the printer working, then I'll work to get the PR completely up-to-date, fix the bed model, and potentially add more nozzle sizes

AC5SHOT, 2024-05-06:
will give it a shot when its ready, if you want he;p all i need is a link too where i can help list the extra nozzles/materials

AC5SHOT, 2024-05-06:
its working perfectly just dialing in my personal settings but you did a outstanding job

cochcoder, 2024-05-06:
Yay, let me know if you find any changes that could be added to the optimal profile! If you upload a screenshot of all the different nozzle settings from the FlashForge slicer, that would be awsome!

cochcoder, 2024-05-06:
Here is a new build that fixes the build plate model and updates the PR

AC5SHOT, 2024-05-07:
> Here is a new [build](https://github.com/cochcoder/OrcaSlicer/actions/runs/8977273239) that fixes the build plate model and updates the PR

i know this sounds odd but now it will  do the actual print motion without dispensing any material 

the only material that dispenses is on the tram line

cochcoder, 2024-05-07:
> > Here is a new [build](https://github.com/cochcoder/OrcaSlicer/actions/runs/8977273239) that fixes the build plate model and updates the PR
> 
> 
> 
> i know this sounds odd but now it will  do the actual print motion without dispensing any material 
> 
> 
> 
> the only material that dispenses is on the tram line

Did it dispense material before this build?

AC5SHOT, 2024-05-07:
yup
and i didnt see any code change between builds for material

AC5SHOT, 2024-05-07:
but its only the material on the model itself everything else is 100 this build

cochcoder, 2024-05-07:
> yup
> 
> and i didnt see any code change between builds for material
> 
> 

Try removing the profile, restart the program, select the profile, and restart the program again. This sometimes works for me, but if it doesn't I'll look into it.

cochcoder, 2024-05-07:
So it fails to extrude or the issue is with the slicer's build plate?

AC5SHOT, 2024-05-07:
it fails to extrude but only the model

the pre extrusion tram ( that front line extrudes) perfectly

tried what you suggested that didnt work 

and also tried it on mac, also didnt work 

even attempted to turn flow to max and all retractions off but for some reason nothing except the tram line extrudes

AC5SHOT, 2024-05-07:
so definitely a new extrusion issue

AC5SHOT, 2024-05-08:
[image]

this is a image i got to try show "line" 

after that nothing extrudes

cochcoder, 2024-05-10:
I can't see anything that would go wrong in the G-Code on my side. I updated the PR and there is a new build [here](https://github.com/cochcoder/OrcaSlicer/actions/runs/9034750553) in case that fixes it. If it doesn't, please send me your project with the issue.

AC5SHOT, 2024-05-12:
Hey ran the latest build and still having the same issue 

also asking if its possible could you change the x origin to 140 to make it centered, 
support multi bed type by default,change the layer height min to 0.02 and the max to 0.8 , Pleas set the default speed limitation on the X and the Y to be 200 mm/s, as well as a few process temple changes to make it more reliable/ true to original slicer presets (sorry for pestering) 

i am also gona include a benchy g code sliced in orca that has that material issue and a benchy g code sliced in flashforge to help you diagnose 

will also add screenshots of the whole profile that i recommend be used in order to insure no printer damage for whoever else wish to use the guider in orca

the link to them is bellow as it wont let me upload them in this repo

https://github.com/AC5SHOT/example-g-code-and-machine-safe-settings

(really sorry for all the pestering/issues)

AC5SHOT, 2024-05-12:
if i had to venture a guess i believe its to do with maybe having to include a set extrusion ration command before each layer change

cochcoder, 2024-05-15:
> (really sorry for all the pestering/issues)

Don't worry about it!

> also asking if its possible could you change the x origin to 140 to make it centered,
support multi bed type by default,change the layer height min to 0.02 and the max to 0.8 , Pleas set the default speed limitation on the X and the Y to be 200 mm/s, as well as a few process temple changes to make it more reliable/ true to original slicer presets (sorry for pestering)

These changes have been implemented. What settings don't match with the original slicer? 

> if i had to venture a guess i believe its to do with maybe having to include a set extrusion ration command before each layer change

If we remove that, then we'll need to use absolute positioning, which tends to not work that well with complex prints. I think the issue was that Orca Slicer was slicing with relative positioning while the start G-Code had the printer use absolute positioning. 

Here is a new [build](https://github.com/cochcoder/OrcaSlicer/actions/runs/9100629331) with the changes, hopefully it works!

AC5SHOT, 2024-05-16:
hey just tried the new build and after the machine completes the pre print auto leveling it immediately begins ramming into the wall and bashing itself when the print starts, whilst having the bed at a random layer.
 I did find this preview looks odd (attached image with the odd green extruder movement), and a image of how the printer sits whyen the bashing starts
[image]
[image]

AC5SHOT, 2024-05-16:
> Don't worry about it!
>
>
> These changes have been implemented. What settings don't match with the original slicer?
>
>
> If we remove that, then we'll need to use absolute positioning, which tends to not [...]

the acceleration speeds would need a 0 taken off of each of them ie 20000 - 2000

cochcoder, 2024-05-17:
> hey just tried the new build and after the machine completes the pre print auto leveling it immediately begins ramming into the wall and bashing itself when the print starts, whilst having the bed at a random layer.
I did find this preview looks odd (attached image with the odd green extruder movement), and a image of how the printer sits whyen the bashing starts

I'll look into potential reasons for why the printer and G-Code seem to acting that way. In the meantime, can you measure the clearance radius around the extruder, the distance from the nozzle's tip to the lower rod, and the distance from the nozzle's tip to the lid in millimeters? This should help with this issue and also prevent us from running into the issue of the nozzle knocking overprints in the future. Side note, if you can't get an accurate reading, rather have extra than less. 

> the acceleration speeds would need a 0 taken off of each of them ie 20000 - 2000 

Sorry about that! We could also bring down the speed entirely if the bashing problem persists, so we don't break anything. 

Did it still extrude the purge line properly or did it not work after the latest build?

AC5SHOT, 2024-05-17:
23 mm to rod from tip
40 mm around nozzle
nozzle tip to gantry 70 mm (would not measure lid its just crazy height)

np i dialed it down before doing the tests its defiantly a software thing , its almost like it gets told to start moving and never stops

after the latest build it dosnt work at all

cochcoder, 2024-05-18:
> after the latest build it dose not work at all

The purge line right?

AC5SHOT, 2024-05-18:
yip both the purge and print didnt work just continuously attempting to ram out the printer

cochcoder, 2024-05-20:
I added the nozzle clearence information and switched back to absolute positioning. Hopefully, this will give a better result, the only issue I see in the slicer is the purge line going through the middle of the build plate. Here is the new [build](https://github.com/cochcoder/OrcaSlicer/actions/runs/9162141096), let me know where the purge line appears and if anything improves.

AC5SHOT, 2024-05-24:
hey i have been testing the changes

all that i had to do on my end was lower the speeds a little bit other than that it works basically like the original slicer it does everything as expected, even run out of filament pausing (the only difference is it doesn't have a preview pic but thats expected)

all that would need to change before the final roll out is the speeds ( i have attached pics of how far i am able to push the machine and still keep it perfectly reliable )

thanks so much you are a absolute gigachad

[image]
[image]
[image]
[image]
[image]

cochcoder, 2024-05-24:
> all that i had to do on my end was lower the speeds a little bit other than that it works basically like the original slicer it does everything as expected, even run out of filament pausing (the only difference is it doesn't have a preview pic but thats expected)

That's awesome! What speeds in particular did you have to lower? Do you know what size the preview pictures are? I may be able to add them if I know their size, i.e. 140x126, it is a bit strange as 140x110 thumbnails are included. 

> all that would need to change before the final roll out is the speeds ( i have attached pics of how far i am able to push the machine and still keep it perfectly reliable )

What quality do you get with these speeds, what speeds are they? Do you have a specific profile in mind they should be used for, or should these be used for the machine's max? 

> thanks so much you are a absolute gigachad

Just paying my dues to a amazing slicer ;)

AC5SHOT, 2024-05-26:
to be true top the original slicer the profiles would be 

standard:---------------------------------------
 0.18mm layer height with a 0,27mm base
base print speed  60mm
travel 80mm
minimum 5mm
first layer 10mm
fist layer max travel 70mm
fist few layer max 30mm
slow down for overhangs yes
max when >75% - 50mm
max when >50% - 80mm
max when >25% - 100mm
max when >10% - 100mm
max when enable overhang infill  100mm

exterior 50% ( 30mm)
exterior max 40mm
visible interior 70%
visible interior max 200mm
vcisible interior 100%
interior max 200mm

fast print settings:-------------------------------------------
standard: 0.3mm layer height with a 0,3mm base
base print speed  80mm
travel 100mm
minimum 5mm
first layer 10mm
fist layer max travel 70mm
fist few layer max 30mm
slow down for overhangs yes
max when >75% - 50mm
max when >50% - 80mm
max when >25% - 100mm
max when >10% - 100mm
max when enable overhang infill  100mm

exterior 50% ( 40mm)
exterior max 40mm
visible interior 70%
visible interior max 200mm
vcisible interior 100%
interior max 200mm

fine print settings:---------------------------------------
standard: 0.12mm layer height with a 0,2mm base
base print speed  50mm
travel 70mm
minimum 5mm
first layer 10mm
fist layer max travel 70mm
fist few layer max 30mm
slow down for overhangs yes
max when >75% - 50mm
max when >50% - 80mm
max when >25% - 100mm
max when >10% - 100mm
max when enable overhang infill  100mm

exterior 50% ( 25mm)
exterior max 40mm
visible interior 70%
visible interior max 200mm
vcisible interior 100%
interior max 200mm

my optimized settings:
layer height 0,12mm
first layer 0,2mm
first layer 10mm
first layer infill 10mm
outer wall 25mm
inner 60mm
sparse 200mm
internal solid 200
top surface 100mm
gap infill 30mm
support 80mm

overhang speeds 

10-25% - 100mm
25-50% - 100mm
50-75%  - 80mm
75-100% - 50mm
bridge
25 mm external 
150% internal

/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

these are all the acceleration /travel speeds for all the profiles
travel speed 200mm

acceleration
normal printing 200mm
outer wall 200mm
inner wall 2000mm
bridge 50%
sparse 100%
internal solid 100%
first layer 200mm
top surface 200mm
travel 2000mm

others 
skirt speed 10

i believe that the machine settings override the 2000mm inner print walls that you mentioned to the printers max settings keeping it safe 

i genuinely dont know what size the preview pics are but given that the default a slicer wont provide a preview pic when exported as gcode and only a preview pic with gxcode

hope this helped otherwise you could maybe direct me to where i could add the, for you 

I DID PERSONALLY TEST THESE SETTING IN ORCA JUST TO MAKE SURE THEY ARE ALL SAFE AND HAVE NO ISSUES

cochcoder, 2024-05-26:
> the default a slicer wont provide a preview pic when exported as gcode and only a preview pic with gxcode

Hmm, seems like the preview picture only works with a proprietary file that Orca Slicer doesn't have...

Also, I'll take a look at the speeds and change them accordingly, I'm hesitant with the 2000 speeds though. Does it actually reach 2000 mm/s or does the machine itself not allow it to go up to that speed? If it doesn't, I think I'll leave it at 200 like most of the speeds.

cochcoder, 2024-05-26:
> I DID PERSONALLY TEST THESE SETTING IN ORCA JUST TO MAKE SURE THEY ARE ALL SAFE AND HAVE NO ISSUES 

Thank you for that!

AC5SHOT, 2024-05-27:
The settings we have in the extruder is capping it at 200 I believe so I just setting it at 200 should be perfect

cochcoder, 2024-05-27:
Thanks for the clarification and all of your help! I'll mark the PR as ready to merge ASAP!

AC5SHOT, 2024-05-27:
Seriously you are AWESOME thanks

[6 more replies]