OrcaSlicer/OrcaSlicer #4593: Chamber Temperature - not setting

Gboy1995, 2024-03-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0 beta

### Operating System (OS)

Windows

### OS Version

11

### Printer

Criality K1

### How to reproduce

1. Filament Settings
2. Filament
3. Print Chamber Temp
4. Chamber Temperature = 50C
5. Active temperature control - un check

[image]

### Actual results

Nothing change in the fluidd screen considering the chamber temp target. The G code didn`t have the line M141 S50.

### Expected results

Insert in the G Code M141 S50 line, and set the chamer temperature target to 50C with active chamber temp control off.

### Project file & Debug log uploads

[TAMPA_DOC_BASE_ABS_28m47s.zip](https://github.com/SoftFever/OrcaSlicer/files/14659060/TAMPA_DOC_BASE_ABS_28m47s.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

TeraCorp, 2024-04-07:
This is still a bug in the 2.0 final release.

dingausmwald, 2024-07-07:
Still in 2.1.1

RWhitney7, 2024-10-04:
Still waiting on this, any update?  I've seen somewhere that adding something like "M141 S[chamber_temperature] ; set chamber temperature manually " in the machine start gcode may be a work around, with "chamber_temperature" variable name possibly being different from printer to printer.

RWhitney7, 2025-01-03:
> Still waiting on this, any update? I've seen somewhere that adding something like "M141 S[chamber_temperature] ; set chamber temperature manually " in the machine start gcode may be a work around, with "chamber_temperature" variable name possibly being different from printer to printer.

Not sure if this was addressed in the latest update, but the solution above worked like a charm

Hamberthm, 2025-04-13:
I have this problem on K1C and Orca 2.2.0. No matter what temperature I configure it will always be something like 35 degrees. When printing ABS every time I need to go to the mainsail page and change the target temperature to 60 after the printing started.