OrcaSlicer/OrcaSlicer #4596: Fan speed threshold not respecting set values in regards to layer times

lhndo, 2024-03-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

LH Stinger

### How to reproduce

The fan speed thresholds are not respected, even if the layer times are below the set limits.

Layer time
[image]

Fan speed
[image]

 Note: there's an additional issue where skirt printing time is not accounted (eg. using tall skirts as draft shields)

### Actual results

Fan kicks in even if layer time is lower than the set threshold

### Expected results

Fan speed thresholds to be respected

### Project file & Debug log uploads

[fan speed.zip](https://github.com/SoftFever/OrcaSlicer/files/14664883/fan.speed.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DOT-Nobody, 2024-03-21:
[image]

same problem

lhndo, 2024-06-20:
Still an issue in 2.1.0-beta

lokertr, 2024-06-30:
Particularly problematic and annoying when you use remote cooling...

kayboy-bebop, 2024-08-21:
[image]
Still a problem in 2.1.1.

kingnoob1377, 2024-10-03:
[image]
[image]
Current method of setting fan speed is broken in many ways and limited even without bugs

kayboy-bebop, 2024-12-15:
Issue still present in 2.2 Linux build. In my case turning off small area flow compensation stopped the error where it randombly drops to the minimum fan speed despite not being at the max time threshold (as shown in previous message), but testing OP's file yeilds the same incorrect fan speed threshold regardless of SAFC being activated or not.

> ![fan_speed_bug_2 1](https://private-user-images.githubusercontent.com/115389530/359780505-2e4c3999-1fc1-4237-adda-24a15b7704a8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyNjc5ODAsIm5iZiI6MTczNDI2NzY4MCwicGF0aCI6Ii8xMTUzODk1MzAvMzU5NzgwNTA1LTJlNGMzOTk5LTFmYzEtNDIzNy1hZGRhLTI0YTE1Yjc3MDRhOC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxNVQxMzAxMjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NmI4MWQyOWYzNTU5NWRiNmRhYWMwZGRkMzRhMGVjMDE2NzdmZGMzMGQ3YjFkZGIyYTBjNjFkYTIxOWU4NjU4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.mMBZMsRSdKxY7GFgoK0HwbLZCl_Y6Rg7axcKRCGM9zY) Still a problem in 2.1.1.

closed 2025-03-23 by github-actions[bot]