OrcaSlicer/OrcaSlicer #4597: Flat overhang begins with outmost line in midair

Rickthebig, 2024-03-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

DIY BLV Cube, created from Troodon as template

### How to reproduce

1. Slice the model in the project file
[image]

2. Inspect layer 96
[image]

### Actual results

While trying to build a flat overhang (technically, it is a bridge), the path taken uses the outmost line in midair, which will inevitably fail.

Please note: Reorienting the model is not an option for specific reasons. This joint is an isolated test part of a larger model, which already is oriented optimally.

### Expected results

The slicer should begin with the shorter path adjacent to the anchor points. Or it should use arc overhang, if it is implemented. This case cries out for arc overhang.

### Project file & Debug log uploads

[Carl_the_Cubeish_Test_Joint.zip](https://github.com/SoftFever/OrcaSlicer/files/14665397/Carl_the_Cubeish_Test_Joint.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Is there a way to use arc overhang, at least as a postprocessor?
It would be an effective workaround.

vgdh, 2024-03-20:
Why you don't to use a tree support?

Rickthebig, 2024-03-20:
> Why you don't to use a tree support?

I tried that but it did not work out well. The support material adds excess garbage that makes loosing the in-place-articulation almost impossible (it is a small, narrow space). I even changed material from PETG to PLA which is less sticky to itself. You can get it done depending on your craftmanship and patience. The point is, there are about a dozen such joints in the model, which turns this enterprise into a nightmare. Beginning with the fact that I try to avoid support as much as possible -  it is a mess!

vgdh, 2024-03-20:
I agree the native support would be super cool
But for now you can definitely use it as a post processor [from this repo](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration)
They have a detail instruction how to use it as post processor.

Rickthebig, 2024-03-21:
I followed instructions per
https://github.com/Kelsch/arc-overhang-orcaslicer-integration
updated python, installed needed libraries, and tried to run the script - but then:

 File "softfever_slicer_post_processing_script.py", line 325, in <module>
    def getPtfromCmd(line:str)->Point | None:
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'

So before the script actually engages interpreting the gcode, it fails apparently due to a syntax error.

GRRRR

vgdh, 2024-03-21:
I had read the code and it starts parse the gcode file and can't find nozzle diameter in settings section so it throw an error.
The code need some tuning for work properly.

vgdh, 2024-03-21:
I made an [issue](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/issues/74 ) for this problem. May be owner fix it.

vgdh, 2024-03-21:
How you will consider to print the internal bone? It think it will definitely stick to the base model.
May be it would be simpler to cut the print at half and print separately a bone and half parts of the sides.
[image]
[image]

Rickthebig, 2024-03-21:
> How you will consider to print the internal bone? It think it will definitely stick to the base model.

Yes but it can be "unsticked" - but after carefully postprocessing it manually. Can take an hour to my experience.

Printing in halves is also an idea to consider. Due to the number of joints this could be also a tedious work. Need to test it.

closed 2024-06-27 by github-actions[bot]