OrcaSlicer/OrcaSlicer #4599: Speed value not accepting "0"

dsplendido, 2024-03-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Ratrig v1 with Fly CDY V3

### How to reproduce

1. Enter value "0" in any layer speed parameter
2. Press enter

### Actual results

[image]

### Expected results

This screenshot shows that values of "0" were acceptable as parameters in a prior version (1.7). This allows Max Volumetric Flow feature to control speed parameters. Values of 1 or greater in Layer Speed parameters overrides the "Autospeed" values. Result is there is no Autospeed related to Max Volumetric Flow operating.
[image]

### Project file & Debug log uploads

[Max volumetric speed test 105-125.zip](https://github.com/SoftFever/OrcaSlicer/files/14668032/Max.volumetric.speed.test.105-125.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

bitingsock, 2024-04-22:
For me, setting it to something like 9999 is the same as 0. Max flow still works.

dsplendido, 2024-04-23:
That did the trick, thanks for the insight. Max flow or "autospeed" is a critical setting for me, coming from SuperSlicer.

closed 2024-04-23 by dsplendido