OrcaSlicer/OrcaSlicer #4601: Add "Random" to Seam Painting

cubitik, 2024-03-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It would be good if we had a tick in seam painting menu named "Random"

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It should randomize the seams in painted area just a little bit, just like in GalaxySlicer (added image). But with a tick named "Random" to enable.
[image]

buzzhuzz, 2024-03-22:
It is already there.

Just paint seams with a bigger *and* select Random seams.

[image]
[image]

vgdh, 2024-03-22:
should be closed, the feature is already there (v 2.0)

cubitik, 2024-03-24:
> should be closed, the feature is already there (v 2.0)

There is no any feature as explained in the suggestion. Also, if you mean the "brush size", seams are just going at the edge of painted area.
[image]
[image]

vgdh, 2024-03-24:
> > should be closed, the feature is already there (v 2.0)
> 
> There is no any feature as explained in the suggestion. Also, if you mean the "brush size", seams are just going at the edge of painted area.

what version of slicer do you use?

cubitik, 2024-03-24:
> > > should be closed, the feature is already there (v 2.0)
> > 
> > There is no any feature as explained in the suggestion. Also, if you mean the "brush size", seams are just going at the edge of painted area.
> 
> what version of slicer do you use? 

Using latest 2.0.0 RC. [image]

[image]

cubitik, 2024-03-24:
> It is already there.
> 
> Just paint seams with a bigger *and* select Random seams.
> 
> [image]
> [image]
> 
> 

This causes all the seams to fall apart randomly. The purpose of the feature I propose is to select the "random" option in the seam dyeing menu and only the painted areas will be random.

closed 2024-06-30 by github-actions[bot]