OrcaSlicer/OrcaSlicer #4605: There is no glyph for dragging embossed text on object in prepare screen

jasonconstant33, 2024-03-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

WIndows 10

### Additional system information

I am doing a project where i am Embossing unique text on many of the same part on the same plate. EG: label 1, label 2, etc. The default placement of the text is no at the center of my parts in the XY direction, so I need to move each one slightly in XY to fix, not a big deal. The issue is that if you look here the text rotation has this wonderful square glyph to select to make sure you will only rotate the text and not change the location of the part on the plate, but the text does not. This results in you 95% of the time moving the part that is being embossed instead of the embossed text, even though the software is clearly in a submenu for editing embossed text. Please FIX. Disable the ability to move parts on the plate in this context or  make a square or spherical glyph where you need to put your mouse to drag the location of the text.

[image]

### Printer

Bambu lab X1C

### How to reproduce

Emboss text and then try to move it without moving the part that is being embossed.

### Actual results

The part moves by accident and this messed up my plate orginization.

### Expected results

You are able to move the embossed text.

### Project file & Debug log uploads

You do not need a file, use a primitive or literally any file in Orca.

[Mom's Plant Labels - Copy.zip](https://github.com/SoftFever/OrcaSlicer/files/14672405/Mom.s.Plant.Labels.-.Copy.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

buzzhuzz, 2024-03-28:
With the emboss tool enabled, text could be moved by keyboard arrows:

1. Arrows - 10mm increment
2. Shift+arrows - 1mm increment.

Hope that helps.

I think it should be helpful to put some tooltips on the screen. Adding a small action cube for translation at the center may be helpful too.

jasonconstant33, 2024-03-29:
I love that feature.But can’t you see that it switching to the part when you drag by accident is a big. I found that it works better when you Aren't looking at your part from a view cube cardianial direction. Why not put a glyph on it to make it work as easily as the rotate. It’s inconsistent functionality; the rotate is fool proof and the drag to move is difficult to use at best.Jason Constant

Arrows - 10mm increment

closed 2024-07-05 by github-actions[bot]