OrcaSlicer/OrcaSlicer #4611: Error: non-manifold edges on standart calibration objects

onlineapps-cloud, 2024-03-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.00 beta

### Operating System (OS)

Linux

### OS Version

Debian 11

### Printer

Anycubic kobra 2 max

### How to reproduce

load anycubic profile
click calibrate in orca slicer
select temperature
catch error

### Actual results

[image]

### Expected results

i expected to not get this error, as i understand integrated 3d models don't need to have errors.

### Project file & Debug log uploads

2024-03-21 11:10:32.240511[Thread 0x00007fc5fef16c80]:gui mode, Current OrcaSlicer Version 2.0.0-beta
2024-03-21 11:10:32.260975[Thread 0x00007fc5fef16c80]:load_language: language , initial: 1
2024-03-21 11:10:32.261020[Thread 0x00007fc5fef16c80]:language provided by OrcaSlicer.conf: en_US
2024-03-21 11:10:32.277827[Thread 0x00007fc5fef16c80]:Update_dark_mode_flag: switch the current dark mode status to 0 
2024-03-21 11:10:32.278926[Thread 0x00007fc5fef16c80]:begin to show the splash screen...
2024-03-21 11:10:32.376875[Thread 0x00007fc5fef16c80]:loading systen presets...
2024-03-21 11:10:32.381428[Thread 0x00007fc5fef16c80]:[Orca Updater]:Checking whether the profile from resource is newer
2024-03-21 11:10:32.389560[Thread 0x00007fc5fef16c80]:initialize_network_module, error, dlerror is /home/user/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
2024-03-21 11:10:32.389612[Thread 0x00007fc5fef16c80]:initialize_network_module, can not Load Library for /home/user/.config/OrcaSlicer/plugins/libbambu_networking.so
2024-03-21 11:10:32.389624[Thread 0x00007fc5fef16c80]:on_init_network: on_init_network, load dll failed
2024-03-21 11:10:32.389633[Thread 0x00007fc5fef16c80]:unload_network_module, network module 0
2024-03-21 11:10:32.389648[Thread 0x00007fc5fef16c80]:on_init_network, unload_network_module, result = 0
2024-03-21 11:10:32.389656[Thread 0x00007fc5fef16c80]:preset_folder: set to empty
2024-03-21 11:10:32.389757[Thread 0x00007fc5fef16c80]:load_presets enter, substitution_rule 3, preferred printer_model_id 
2024-03-21 11:10:32.389787[Thread 0x00007fc5fef16c80]:load_vendor_configs_from_json enter, path /home/user/.config/OrcaSlicer/system, compatibility_rule 2
2024-03-21 11:10:32.392036[Thread 0x00007fc5fef16c80]:load_vendor_configs_from_json: parse /home/user/.config/OrcaSlicer/system/Anycubic.json, got description:  "Anycubic configurations"
2024-03-21 11:10:32.397515[Thread 0x00007fc5fef16c80]:operator() 35100.15mm Optimal @Anycubic i3MegaS load filament_id: 
2024-03-21 11:10:32.398282[Thread 0x00007fc5fef16c80]:operator() 35100.15mm Optimal @Anycubic Chiron load filament_id: 
2024-03-21 11:10:32.398973[Thread 0x00007fc5fef16c80]:operator() 35100.15mm Optimal @Anycubic Vyper load filament_id: 
2024-03-21 11:10:32.399578[Thread 0x00007fc5fef16c80]:operator() 35100.15mm Optimal @Anycubic KobraMax load filament_id: 
2024-03-21 11:10:32.400185[Thread 0x00007fc5fef16c80]:operator() 35100.15mm Optimal @Anycubic KobraPlus load filament_id: 
2024-03-21 11:10:32.400793[Thread 0x00007fc5fef16c80]:operator() 35100.15mm Optimal @Anycubic 4MaxPro2 load filament_id: 
2024-03-21 11:10:32.401404[Thread 0x00007fc5fef16c80]:operator() 35100.15mm Optimal @Anycubic Kobra2 load filament_id: 
2024-03-21 11:10:32.402068[Thread 0x00007fc5fef16c80]:operator() 35100.20mm Standard @Anycubic i3MegaS load filament_id: 
2024-03-21 11:10:32.402736[Thread 0x00007fc5fef16c80]:operator() 35100.20mm Standard @Anycubic Chiron load filament_id: 
2024-03-21 11:10:32.403480[Thread 0x00007fc5fef16c80]:operator() 35100.20mm Standard @Anycubic Vyper load filament_id: 
2024-03-21 11:10:32.404152[Thread 0x00007fc5fef16c80]:operator() 35100.20mm Standard @Anycubic KobraMax load filament_id: 
2024-03-21 11:10:32.404832[Thread 0x00007fc5fef16c80]:operator() 35100.20mm Standard @Anycubic KobraPlus load filament_id: 
2024-03-21 11:10:32.405204[Thread 0x00007fc5fef16c80]:operator() 35100.20mm Standard @Anycubic 4MaxPro load filament_id: 
2024-03-21 11:10:32.405960[Thread 0x00007fc5fef16c80]:operator() 35100.20mm Standard @Anycubic 4MaxPro2 load filament_id: 
2024-03-21 11:10:32.406752[Thread 0x00007fc5fef16c80]:operator() 35100.20mm Standard @Anycubic Kobra2 load filament_id: 
2024-03-21 11:10:32.407425[Thread 0x00007fc5fef16c80]:operator() 35100.30mm Draft @Anycubic i3MegaS load filament_id: 
2024-03-21 11:10:32.408091[Thread 0x00007fc5fef16c80]:operator() 35100.30mm Draft @Anycubic Chiron load filament_id: 
[... 401 lines cut ...]
2024-03-21 11:11:30.644814[Thread 0x00007fc5fef16c80]:update_background_process, Line 5030: was_running = 0, running 0, invalidated=1, return_state=4, internal_cancel=0
2024-03-21 11:11:30.644823[Thread 0x00007fc5fef16c80]:get_enable_print_status: m_print_select 3, enable= 0 
2024-03-21 11:11:30.644826[Thread 0x00007fc5fef16c80]:update_slice_print_status m_slice_select 1: can_slice= 0, can_print 1, enable_slice 0, enable_print 0 
2024-03-21 11:11:30.644884[Thread 0x00007fc5fef16c80]:update_background_process, Line 5050: set to process_completed_with_error, return_state=4
2024-03-21 11:11:30.644889[Thread 0x00007fc5fef16c80]:update_background_process, Line 5103: exit, return_state=4
2024-03-21 11:11:30.644893[Thread 0x00007fc5fef16c80]:restart_background_process, Line 5141: not started
2024-03-21 11:11:33.570003[Thread 0x00007fc5fd68c700]:process_task{ type:AddObject, id:2, path:/tmp/orcaslicer_model/Thu_Mar_21/11_11_29#14648#55, object:131, extra:1} and interval = 10
2024-03-21 11:11:33.570088[Thread 0x00007fc5fd68c700]:_add_model_file_to_archive:6290, filename 3D/Objects/temperature_tower.stl_2.model, m_split_model 0,  sub_model 1
2024-03-21 11:11:33.570124[Thread 0x00007fc5fd68c700]:bbs_3mf: save key= BambuStudio:3mfVersion, value = 1
2024-03-21 11:11:33.849543[Thread 0x00007fc5fef16c80]:process_ui_task{ type:AddObject, id:2, path:/tmp/orcaslicer_model/Thu_Mar_21/11_11_29#14648#55, object:131, extra:1} and interval = 10
2024-03-21 11:11:36.849627[Thread 0x00007fc5fd68c700]:process_task{ type:AddObject, id:3, path:/tmp/orcaslicer_model/Thu_Mar_21/11_11_29#14648#55, object:183, extra:1} and interval = 10
2024-03-21 11:11:36.849868[Thread 0x00007fc5fd68c700]:_add_model_file_to_archive:6290, filename 3D/Objects/temperature_tower.stl_3.model, m_split_model 0,  sub_model 1
2024-03-21 11:11:36.849986[Thread 0x00007fc5fd68c700]:bbs_3mf: save key= BambuStudio:3mfVersion, value = 1
2024-03-21 11:11:36.928870[Thread 0x00007fc5fd68c700]:process_task{ type:Backup, id:0, path:, object:0, extra:11} and interval = 10
2024-03-21 11:11:36.928914[Thread 0x00007fc5fef16c80]:process_ui_task{ type:AddObject, id:3, path:/tmp/orcaslicer_model/Thu_Mar_21/11_11_29#14648#55, object:183, extra:1} and interval = 10
2024-03-21 11:11:36.928942[Thread 0x00007fc5fef16c80]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:11} and interval = 10
2024-03-21 11:11:36.930848[Thread 0x00007fc5fef16c80]:export_3mf: path=, backup=1, export_plate_idx=-1, SaveStrategy=69748
2024-03-21 11:11:36.930880[Thread 0x00007fc5fef16c80]:store_to_3mf_structure: plate 1 before load, width 0, height 0, size 0!
2024-03-21 11:11:36.930886[Thread 0x00007fc5fef16c80]:store_to_3mf_structure: plate 1 after load, width 0, height 0, size 0!
2024-03-21 11:11:36.930891[Thread 0x00007fc5fef16c80]:store_to_3mf_structure: plate 0, gcode_filename=, with_slice_info=1, slice_valid 0, object item count 1.

### Checklist of files to include

- [X] Log file
- [ ] Project file

onlineapps-cloud, 2024-03-21:
this happens for all calibration models that i tried to open

CLNilix, 2024-04-30:
The bug still exists in release version 2.0.0.
i can repair it with Windows tools, but that shouldn't be intended way for a calibration tool .

[image]

The-Monkey-King, 2024-05-05:
Log file alert:
```
[error]	2024-05-05 17:34:07.760129[Thread 0x0000258c]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-05-05 17:34:11.495805[Thread 0x0000258c]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
```

closed 2024-08-11 by github-actions[bot]