OrcaSlicer/OrcaSlicer #4614: Edit Custom G-Code Window Too Large for Window

JemC42, 2024-03-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Creality Ender 3 S1 Pro

### How to reproduce

1. Edit Printer preset
2. Select Machin G-Code tab
3. Click Edit Custom G-Code button under the Machine End G-code / Machine Start G-Code
4. Edit pop up window opens but is not sized for the screen. Its too large in vertical dimension

### Actual results

window opens but is not sized for the screen. Its too large in vertical dimension
[image]

### Expected results

would expect the pop up window to be sized to fit the screen and be resizeable

### Project file & Debug log uploads

[Untitled.zip](https://github.com/SoftFever/OrcaSlicer/files/14695673/Untitled.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

foreachthing, 2024-03-22:
Ooooh.... why is this window fix in size? 👎🏼 
This window should be sizeable ☝🏼  !

LindyBalboa, 2024-06-16:
Do you have display scaling turned on at the Windows OS level? I had mine at 150%, because 100% with a high DPI monitor is just too small for me, and setting it back to 100% fixes this window.

closed 2024-09-22 by github-actions[bot]