OrcaSlicer/OrcaSlicer #4616: Appimage - error while loading shared libraries: libwebkit2gtk-4.0.so.37

huuxfluux, 2024-03-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer_Linux_V2.0.0-beta.AppImage 

### Operating System (OS)

Linux

### OS Version

MANJARO 6.7.7-1 

### Additional system information

System:
  Kernel: 6.7.7-1-MANJARO arch: x86_64 bits: 64 compiler: gcc v: 13.2.1
    clocksource: tsc avail: hpet,acpi_pm
    parameters: BOOT_IMAGE=/boot/vmlinuz-6.7-x86_64
    root=UUID=8a3d48b0-be62-4c5e-b601-d6852651ba19 rw quiet splash
    udev.log_priority=3
  Desktop: KDE Plasma v: 5.27.11 tk: Qt v: 5.15.12 info: frameworks
    v: 5.115.0 wm: kwin_x11 vt: 2 dm: SDDM Distro: Manjaro base: Arch Linux
Machine:
  Type: Desktop System: Micro-Star product: MS-7A33 v: 2.0
    serial: <superuser required>
  Mobo: Micro-Star model: X370 SLI PLUS (MS-7A33) v: 2.0
    serial: <superuser required> uuid: <superuser required>
    UEFI: American Megatrends v: 3.JU date: 11/02/2021
Battery:
  ID-1: hidpp_battery_0 charge: 78% condition: N/A volts: 4.0 min: N/A
    model: Logitech G903 LIGHTSPEED Wireless Gaming Mouse w/ HERO type: N/A
    serial: <filter> status: discharging
Memory:
  System RAM: total: 32 GiB available: 31.28 GiB used: 6.88 GiB (22.0%)
  Message: For most reliable report, use superuser + dmidecode.
  Array-1: capacity: 128 GiB slots: 4 modules: 4 EC: None
    max-module-size: 32 GiB note: est.
  Device-1: Channel-A DIMM 0 type: DDR4 detail: synchronous unbuffered
    (unregistered) size: 8 GiB speed: 3200 MT/s volts: note: check curr: 1
    min: 1 max: 1 width (bits): data: 64 total: 64 manufacturer: Corsair
    part-no: CMK16GX4M2B3200C16 serial: N/A
  Device-2: Channel-A DIMM 1 type: DDR4 detail: synchronous unbuffered
    (unregistered) size: 8 GiB speed: 3200 MT/s volts: note: check curr: 1
    min: 1 max: 1 width (bits): data: 64 total: 64 manufacturer: Corsair
    part-no: CMK16GX4M2B3200C16 serial: N/A
  Device-3: Channel-B DIMM 0 type: DDR4 detail: synchronous unbuffered
    (unregistered) size: 8 GiB speed: 3200 MT/s volts: note: check curr: 1
    min: 1 max: 1 width (bits): data: 64 total: 64 manufacturer: Corsair
    part-no: CMK16GX4M2B3200C16 serial: N/A
  Device-4: Channel-B DIMM 1 type: DDR4 detail: synchronous unbuffered
    (unregistered) size: 8 GiB speed: 3200 MT/s volts: note: check curr: 1
    min: 1 max: 1 width (bits): data: 64 total: 64 manufacturer: Corsair
    part-no: CMK16GX4M2B3200C16 serial: N/A
PCI Slots:
  Permissions: Unable to run dmidecode. Root privileges required.
CPU:
  Info: model: AMD Ryzen 5 3600X bits: 64 type: MT MCP arch: Zen 2 gen: 3
    level: v3 note: check built: 2020-22 process: TSMC n7 (7nm) family: 0x17 (23)
    model-id: 0x71 (113) stepping: 0 microcode: 0x8701021
  Topology: cpus: 1x cores: 6 tpc: 2 threads: 12 smt: enabled cache:
    L1: 384 KiB desc: d-6x32 KiB; i-6x32 KiB L2: 3 MiB desc: 6x512 KiB L3: 32 MiB
    desc: 2x16 MiB
  Speed (MHz): avg: 2200 min/max: 2200/4409 boost: enabled scaling:
    driver: acpi-cpufreq governor: schedutil cores: 1: 2200 2: 2200 3: 2200
    4: 2200 5: 2200 6: 2200 7: 2200 8: 2200 9: 2200 10: 2200 11: 2200 12: 2200
    bogomips: 91236
[... 237 lines cut ...]
As far as i understand appimages no libraries outside the appimage itself is supposed to be needed.

### Project file & Debug log uploads

   ~/AppImage  ./OrcaSlicer_Linux_V2.0.0-beta.AppImage                                                            ✔  49s  
/tmp/.mount_OrcaSlVVl9XN/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
    ~/AppImage  pacman -Qs webkit                                                                                      127 ✘ 
local/libwpe 1.14.2-1
    General-purpose library for WPE WebKit
local/webkit2gtk-4.1 2.42.5-2
    Web content engine for GTK
local/wpebackend-fdo 1.14.2-1
    Freedesktop.org backend for WPE WebKit
    ~/AppImage  pacman -Qs libwebkit                                                                                       ✔ 
local/webkit2gtk-4.1 2.42.5-2
    Web content engine for GTK

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

If you need anything of me to test, I'll be willing to.

huuxfluux, 2024-03-21:
I just want to add that i get the exact same problem with `OrcaSlicer_Linux_V1.9.1.AppImage`

cochcoder, 2024-03-21:
Does [this](https://github.com/cochcoder/OrcaSlicer/actions/runs/8378359697/artifacts/1347377967) fix your issue?

huuxfluux, 2024-03-21:
> Does [this](https://github.com/cochcoder/OrcaSlicer/actions/runs/8378359697/artifacts/1347377967) fix your issue?

No.

I installed webkit2gtk with $ pacman -S webkit2gtk
And then it installs, but the program gets all kinds of errors.

[image]
"use system SSL certificate: /etc/ssl/certs/ca-certificates.crt
To manually specify the system certificate store, set the SSL_CERT_FILE environment variable to the correct
CA and restart the application
Do you want to continue?"

First a screenshot of how the app looks:
[image]

And then i'm gonna drop a log from terminal from what i get:
`$ ./OrcaSlicer_Linux_V2.0.0-beta_build.AppImage                                                255 ✘  2m 3s  
[2024-03-21 18:39:42.302373] [0x00007095e6044480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/uxu/.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/uxu/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.622: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:44.632: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.632: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:44.632: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.632: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:44.633: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.633: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:44.633: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.633: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:44.634: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.634: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:44.635: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.635: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:44.635: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.635: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:44.636: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.636: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:44.637: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.637: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:44.637: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:44.637: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
src/nv_gbm.c:288: GBM-DRV error (nv_gbm_create_device_native): nv_common_gbm_create_device failed (ret=-1)

[... 269 lines cut ...]
(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:46.395: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:46.395: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:46.396: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:46.396: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:30168): GLib-GObject-CRITICAL **: 18:39:46.396: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:30168): Gtk-CRITICAL **: 18:39:46.396: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:30168): CRITICAL **: 18:39:46.402: Cannot register URI scheme wxfs more than once

** (orca-slicer:30168): CRITICAL **: 18:39:46.402: Cannot register URI scheme memory more than once
src/nv_gbm.c:288: GBM-DRV error (nv_gbm_create_device_native): nv_common_gbm_create_device failed (ret=-1)

KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 820x632: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 820x632: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 820x632: Permission denied
Failed to create EGL images for DMABufs with file descriptors -1, -1 and -1
`

cochcoder, 2024-03-21:
Just to clarify, installing libwebkit on your system fixes this issue?

huuxfluux, 2024-03-21:
> Just to clarify, installing libwebkit on your system fixes this issue?

It resolved the issue of this error:
"/tmp/.mount_OrcaSl8SlKCj/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory"

But more problems appeared. Maybe make a separate issue?

Isn't the point of appimage to be independent of dependencies?

cochcoder, 2024-03-21:
> But more problems appeared. Maybe make a separate issue?

Make a separate issue if these problems don't go away once this issue is fixed. 

> Isn't the point of appimage to be independent of dependencies?

I'm just making sure that installing libwebkit resolves the issue and that is isn't some other issue that is causing this, as libwebkit is installed with libwebkit2gtk-4.0-dev

cochcoder, 2024-03-21:
Do you think it could potentially be a OS issue? Manjaro doesn't have the best track record at having consistent stable builds. I don't have a issue with the Appimage (I use Gear Lever on Debian)

huuxfluux, 2024-03-21:
> I'm just making sure that installing libwebkit resolves the issue and that is isn't some other issue that is causing this, as libwebkit is installed with libwebkit2gtk-4.0-dev

The package installed was "webkit2gtk" which contains "libwebkit2gtk-4.0.so.37".
This can be found with `sudo pacman -Fyx libwebkit2gtk-4.0.so.3` 

> Do you think it could potentially be a OS issue? Manjaro doesn't have the best track record at having consistent stable builds. I don't have a issue with the Appimage (I use Gear Lever on Debian)

I mean, Cura's appimage runs with no issue at all.
And maybe these appimages should be tested on other **arch based distros**.
Perhaps i could test on another computer with endeavour OS for example.

cochcoder, 2024-03-21:
I'm going to try three things and see if one of them fix the issue, or at least gets us on the right path.

huuxfluux, 2024-03-21:
Ok so i tested the appimage V2.0.0-beta on my laptop that runs Manjaro GNOME instead of KDE Plasma. 
There i ran in to no issue at all!

Installed Endeavour OS on a VM. Same issue again! Guess what, it runs KDE Plasma!
Asked on the Manjaro forum and people who ran other then KDE had no issues.

So it might be a KDE Plasma related issue!

cochcoder, 2024-03-22:
That's strange, I run KDE Plasma 5.27.5 and I have no issues, what version of KDE did you test it on?

cochcoder, 2024-03-22:
Let me know if any of these improve things, thank you for your testing!
 
1) https://github.com/cochcoder/OrcaSlicer/actions/runs/8384746810/artifacts/1348788426

2) https://github.com/cochcoder/OrcaSlicer/actions/runs/8384753862/artifacts/1348789730

huuxfluux, 2024-03-22:
> Let me know if any of these improve things, thank you for your testing!
> 
>     1. https://github.com/cochcoder/OrcaSlicer/actions/runs/8384746810/artifacts/1348788426
> 
>     2. https://github.com/cochcoder/OrcaSlicer/actions/runs/8384753862/artifacts/1348789730

So I've tested around some more. Seems like those 2 appimages runs better for some reason.  Now the program is runs.
With that said. I had to install webkit2gtk on all tries on KDE. Both on KDE Plasma 5.27.11 and 6.0.2.
And also my testing was done on my laptop with both Nvidia and Intel GPU. 
I'm thinking why Ocraslicer won't run on my main PC is because of Nvida, since that is my only GPU on it.

huuxfluux, 2024-03-23:
Ok so i tested some other slicers, bambulabs and prusia. Also your build versions. With webkit2gtk and without installed.
Made a screencapture of my testing if you're interested.
That way you can see what happens for me.
https://youtu.be/gLQiANTVt-U

> I added webkit2gtk-driver to the dependency list in both. In what ways is it better? Send some screenshots if applicable.
> 

So i tested on my laptop with endeavour os KDE Plasma 6. I'll make a new post of some screenshots from there.

huuxfluux, 2024-03-23:
(laptop intel GPU endeavourOS testing)
Ok so i think something went bad with on this one, i downloaded it again and then it worked like the others.
Is there checksums avaible?
[image]

But after i once again installed `webket2gtk ` all 3 versions of V2.0.0beta worked.
[image]

guix77, 2024-03-25:
Hi,

I'm on Debian 12 & KDE 5.27.5, having the issue even with your last link @cochcoder :

`/tmp/.mount_OrcaSliXyKQd/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory`

Did not try to install anything libwebkit2gtk though. But AppImages should not need anything external, right?

cochcoder, 2024-03-25:
> But AppImages should not need anything external, right?

I believe that is correct, I'm pretty sure that the Appimage should contain all the needed dependencies and packages for it to run without anything extra installed on the system. 

Any help in this issue would be greatly appreciated as I tried adding multiple dependencies to the Appimage application and can't find a solution to this problem. Hopefully the flatpak version of Orca Slicer fixes this issue, even though it would've been ideal for the Appimage to work.

guix77, 2024-03-25:
It looks like this issue has been here for a while: https://github.com/SoftFever/OrcaSlicer/issues/425

After installing `sudo apt install libwebkit2gtk-4.0-37` I also confirm @huuxfluux 's results with the certificate in https://github.com/SoftFever/OrcaSlicer/issues/4616#issuecomment-2013154083

cochcoder, 2024-03-25:
> After installing sudo apt install libwebkit2gtk-4.0-37 I also confirm @huuxfluux 's results with the certificate in https://github.com/SoftFever/OrcaSlicer/issues/4616#issuecomment-2013154083

Strange, I did include that in the list of dependencies to install, could it be a issue with how the Appimage is packaged?

guix77, 2024-03-25:
Mmm, I just clicked on Yes and it used the system certificates. So I don't think it's an issue there, although this should be in the stable release doc. It's also mentionned in the Dockerfile so it's probably just there in cases we need to specifiy another certificate file.

guix77, 2024-03-26:
> Strange, I did include that in the list of dependencies to install, could it be a issue with how the Appimage is packaged?

Or a limit of AppImage system? I've read it somewhere but didn't dig.

cochcoder, 2024-03-26:
> Or a limit of AppImage system, being unable to package low level desktop libs? I've read it somewhere but didn't dig.

Interesting... I'll look into it more sometime.

acornejo, 2024-03-28:
AppImage does not include libwebkit or does not work for some reason. Note that SuperSlicer, PrusaSlicer and Cura do not have this issue despite me not having libwebkit installed, so I think this is an issue on what is included in this AppImage.

discip, 2024-03-30:
> ![orca-question](https://private-user-images.githubusercontent.com/77159995/315564642-6ad71e92-9154-4792-b691-77adba2e153d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE4MzQ2MTQsIm5iZiI6MTcxMTgzNDMxNCwicGF0aCI6Ii83NzE1OTk5NS8zMTU1NjQ2NDItNmFkNzFlOTItOTE1NC00NzkyLWI2OTEtNzdhZGJhMmUxNTNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzMwVDIxMzE1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMyNTM3NjU3NDVkODZlYTY2OTNlMzZjYzhmYmY0Y2UyMGVkZDg4NjJjMjQxYWJjOWEyZDUxOTNjNjBmYWJmNjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.LcbF3WDN70C9TFP8DrwCqCwZUcV1Wb-ak-030liLmds) "use system SSL certificate: /etc/ssl/certs/ca-certificates.crt
To manually specify the system certificate store, set the SSL_CERT_FILE environment variable to the correct CA and restart the application
Do you want to continue?"

Same here on Manjaro (Gnome) [**`6.7.7-1`** & **`6.8.0rc6-1`**].

Every time this popup followed by a black splash-screen.

Darnyifox, 2024-04-03:
Absence of webkit library and blank app window is separated issue. Webkit2 compositing fails on some Nvidia. Try run as:
> WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.0.0.AppImage

It works for me. (Same on BambuStudio.)
[https://bugs.webkit.org/show_bug.cgi?id=180739](https://bugs.webkit.org/show_bug.cgi?id=180739)

cochcoder, 2024-04-03:
I have a attempt at a build that automagically applies this [here](https://github.com/cochcoder/OrcaSlicer/actions/runs/8544136079#artifacts), hopefully this works

discip, 2024-04-03:
@cochcoder

Is the change applied system wide after testing your build or is only the AppImage affected?
If it is system wide, how do I revert that?

cochcoder, 2024-04-03:
> @cochcoder
> 
> Is the change applied system wide after testing your build or is only the AppImage affected?
> 
> If it is system wide, how do I revert that?

Based off my limited terminal experience, running `WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.0.0.AppImage` should only apply on that launch and not others. I may be wrong though...

discip, 2024-04-03:
Sorry I didn't make myself clear enough:
I did run the AppImage you provided to test.

Does the change you implemented in that build affect all instances of OrcaSlicer or just the ones I tested?

cochcoder, 2024-04-03:
> Sorry I didn't make myself clear enough:
> 
> I did run the AppImage you provided to test.
> 
> 
> 
> Does the change you implemented in that build affect all instances of OrcaSlicer or just the ones I tested?

Ah, sorry about that, the change I implemented was adding the environment variable to the CI build job. This should only affect the versions of Orca Slicer that have this applied. Some testing by switching between my implementation and the standard orca slicer version would confirm this suspicion. I can't test this as I don't have Nvida GPUs and don't currently know how to check if the environment variable is applied.

discip, 2024-04-03:
Yeah, I am also on an AMD machine, but I'm having this all black splash screen on every start and thought your build might fix this 😅, but unfortunately it didn't.

<details>
  <summary>
    <b>setup</b>
  </summary>
<br>

CPU | GPU | RAM | SSD
-|-|-|-
Ryzen 9 7950X3D | XFX 7900 XTX | G.Skill Flare X5 32 GB | Samsung 990 Pro 2 TB

</details>

discip, 2024-04-03:
No need to be sorry. 👍

Yeah, I am also on an AMD machine, but I'm having this all black splash screen on every start and thought your build might fix this 😅, but unfortunately it didn't.

<details>
  <summary>
    <b>setup</b>
  </summary>
<br>

CPU | GPU | RAM | SSD
-|-|-|-
Ryzen 9 7950X3D | XFX 7900 XTX | G.Skill Flare X5 32 GB | Samsung 990 Pro 2 TB

</details>

cochcoder, 2024-04-03:
> No need to be sorry. 👍
> 
> Yeah, I am also on an AMD machine, but I'm having this all black splash screen on every start and thought your build might fix this 😅, but unfortunately it didn't.
> setup
> 
> CPU 	GPU 	RAM 	SSD
> Ryzen 9 7950X3D 	XFX 7900 XTX 	G.Skill Flare X5 32 GB 	Samsung 990 Pro 2 TB

Nice build! Did the `WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.0.0.AppImage` command fix your issue? (before the build I provided)

discip, 2024-04-03:
Haven't tried it.

Darnyifox, 2024-04-03:
this build still seems broken on Nvidia quadro K5000

cochcoder, 2024-04-03:
@discip Please let me know if using the command `WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.0.0.AppImage` works for you or not, it'll be helpful to know whether the fix or my build doesn't work for you (this would also narrow down what is the cause of your issue if it isn't being fixed by the command)

@Darnyifox did the command fix the issue for you and the build isn't working?

discip, 2024-04-03:
> `WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.0.0.AppImage`

Where do I enter this?

cochcoder, 2024-04-03:
> > `WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.0.0.AppImage`
> 
> Where do I enter this?

In the terminal, once you cd to the folder where Orca Slicer is stored, keep in mind that if it isn't named OrcaSlicer_Linux_V2.0.0.AppImage then you'll want to change it to its current name

Darnyifox, 2024-04-03:
WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.1.0-dev.AppImage    -OK
./OrcaSlicer_Linux_V2.1.0-dev.AppImage -still bad
After going through the initial configuration, the defect is manifested only by the failure to display the home page. To try the black window again, I would probably have to delete the configuration.

discip, 2024-04-03:
In the meantime I figured it out, but unfortunately this doesn't change anything. 🤷🏻

cochcoder, 2024-04-03:
> In the meantime I figured it out, but unfortunately this doesn't change anything. 🤷🏻

It seems like the issue you're experiencing is different than the bug with Webkit and Nvidia

discip, 2024-04-03:
[Screencast from 2024-04-04 00-33-38.webm](https://github.com/SoftFever/OrcaSlicer/assets/53649486/e254d564-8d1a-4666-9667-71040a042fd5)

cochcoder, 2024-04-03:
> Screencast.from.2024-04-04.00-33-38.webm

I have the same issue of this on Debian 12, it sucks, but at least it works

discip, 2024-04-03:
I'm on 6.8.0-1-MANJARO Gnome.

Are you also using an AMD machine?

cochcoder, 2024-04-03:
Intel (iGPU)

discip, 2024-04-03:
In that case the GPU should not be responsible for this issue, right?

cochcoder, 2024-04-03:
> In that case the GPU should not be responsible for this issue, right?

In the case of a hardware issue, no, it's not responsible. In the case of software, I have no clue.

monyarm, 2024-04-07:
This is also an issue on gentoo.
I linked `/usr/lib64/libwebkit2gtk-4.1.so.0` to `/usr/lib64/libwebkit2gtk-4.0.so.37`
And then I get `/tmp/.mount_OrcaSlZbnrLB/bin/orca-slicer: error while loading shared libraries: libjavascriptcoregtk-4.0.so.18: cannot open shared object file: No such file or directory`

After linking it as well, orcaslicer finally started.

AppImages are supposed to contain all the libraries they need, so why not include these two?

Colcobau, 2024-04-09:
Having the same issue on my end. 

Using a Read-Only distro (SteamOS) so I'm unable to install  libwebkit2gtk-4.0.so.37 without jumping through hoops (only for it to go to shit after an update).

I thought AppImages were supposed to include all dependencies built in?

`/tmp/.mount_OrcaSldEdvyH/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
`

jmarmstrong1207, 2024-04-13:
This REALLY needs to be fixed. AppImage was specifically created so that there wouldn't be any dependency issues no matter the distribution. Not adding this dependency is exactly the opposite of what AppImage is intended for.

huuxfluux, 2024-04-13:
> @discip Please let me know if using the command `WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.0.0.AppImage` works for you or not, it'll be helpful to know whether the fix or my build [...]

THIS! 🔓
Worked just fine when i tried on 3 versions!

- OrcaSlicer_Linux_V1.9.1.AppImage

- OrcaSlicer_Linux_V2.0.0-beta.AppImage

- OrcaSlicer_Linux_V2.0.0.AppImage

System, kernel, KDE version and Nvidia driver all the same as when i first reported!

(could have tested earlier sorry for that 🫣)

cochcoder, 2024-04-13:
That's interesting, this solution works well for some and not for others

Colcobau, 2024-04-17:
On my end, I swapped distros to Nobara, and the issue persisted, however, after running:

`dnf install webkit2gtk4.0`

The Appimage launches just fine without any extra arguments.

yadayada, 2024-04-20:
For the upcoming Ubuntu Noble (24.04), the following worked
```
sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18
```
Thanks to @monyarm for the hint.

igorpoubel, 2024-04-20:
I installed libwebkit2gtk-4.0.so.37 and, because my system is in Portuguese (I'm from Brasil), the en_GB locale wasn't installed. So I just intalled it and OrcaSlicer opened. Just commenting because someone could having similar problem.

BTW: I'm using the 2.0 version.

frankbonatelli, 2024-04-20:
> For the upcoming Ubuntu Noble (24.04), the following worked
>
> ```
> sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
> sudo ln -sf [...]

Yup this fixed the appimage launching issue for ubuntu 24.04
thanks!!

eloebl, 2024-04-20:
> For the upcoming Ubuntu Noble (24.04), the following worked
>
> ```
> sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
> sudo ln -sf [...]

Thank you! I also suspect this may fix it long term without the work around; https://github.com/SoftFever/OrcaSlicer/issues/5007

buzzhuzz, 2024-04-23:
Symbolic link to a library with different ABI version work just because of a miracle. 

Looks like distributions started removing older version of libwebkit 4.0 (it's been few years since 4.1 was introduced). 

Because of this correct solution would be to bundle libwebkit library Orca built against into AppImage since it is no longer valid to rely on library presence on users host system.

buzzhuzz, 2024-04-23:
> I also suspect this may fix it long term without the work around; #5007

That change would allow Orca build on Ubuntu 24.04. However, once it is linked to libwebkit2gtk-4.1 it would expect to have this library on users PC as well. That might create problems on older systems like Ubuntu 20.04 which does not have libwebkit2gtk-4.1 in their repos. The only correct solution would be to bundle that library into AppImage.

Another reason to do this is #4986. In that issue user have not had libwebkit2gtk installed altogether which breaks Orca AppImage. Normally that would be sorted out by distribution packaging system (RPM, DEB, pacman), but AppImage do not interact with them.

domsch1988, 2024-04-24:
Just wanted to say that this is an issue on Fedora 40 based ublue images as well. I'm running the latest Aurora image. And it being an immutable distro, having to install additional system libraries is pretty much no what i'd like to do. I hoped an Appimage would take care of libraries.
Any chance this could be distributed as a flatpak? That should also solve this issue.

[37 more replies]

closed 2024-07-01 by SoftFever