OrcaSlicer/OrcaSlicer #4619: Memory Leak + High CPU Load + High GPU Load

Orel-A, 2024-03-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro 23H2

### Additional system information

CPU: Ryzen 7600X
GPU: AMD Radeon RX 5700 XT
RAM: DDR5 32GB 6000MHZ CL30

### Printer

Creality Ender-3 V3 KE

### How to reproduce

Honestly, for now I don't know how to reproduce this, but I'm opening this issue as it happened to me more than once and maybe more people have this problem.
What I do remember is that I opened two instances of Orca at the same time...

### Actual results

Screenshots of Task Manager:
[image]
[image]
[image]

### Expected results

-

### Project file & Debug log uploads

[debug_Thu_Mar_21_18_26_58_310884.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14704416/debug_Thu_Mar_21_18_26_58_310884.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Even after closing all running Orca instances, the memory leak is still present. Meaning I still have around 14GB RAM usage in Windows.

Embeddi, 2024-03-23:
I'm also seeing 100% **GPU** load with 2.0.0-rc1 on the "Preview" tab. That happens even when OrcaSlicer window is minimized.

When switching back to "Prepare" then GPU load goes down to <1% again.

UserDmitry, 2024-03-30:
Same issue here. When in Preview mode and top right menu scrolled down, some CPU cores are at 100% load. When menu is up - no additional cpu usage. 
[image]
When menu is up...
[image]

cochcoder, 2024-04-18:
> It's happening when switching to Color scheme: Speed, in preview:
> 
> ![image](https://private-user-images.githubusercontent.com/7490366/322281505-6a66cfcf-693b-4cdf-8ae6-c17774b1ef68.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM0MDc0MjcsIm5iZiI6MTcxMzQwNzEyNywicGF0aCI6Ii83NDkwMzY2LzMyMjI4MTUwNS02YTY2Y2ZjZi02OTNiLTRjZGYtOGFlNi1jMTc3NzRiMWVmNjgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MThUMDIyNTI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yjg2M2JkZWZkNzI5YmIwYzdmOWFhNzZjMGVmMjE1NTMxNDUwNDg0MmMyOTljNDgyNTExMWM0YmY3ZWU0NzhiMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.t0EGzuQWKXaTY1DBjLFH8-5Z_QgPeqmyZQSz1CGwnMA)
> 
> @SoftFever @cochcoder can you confirm?
> 
> Attaching an example: [Voronoi Pebble Coaster.zip](https://github.com/SoftFever/OrcaSlicer/files/14971123/Voronoi.Pebble.Coaster.zip)

I do notice that more of my CPU cores would be used at higher percentages as well when switching to color, I have also noticed that the interface gets very laggy when view large objects even though the CPU usage is at about 5-10% and Prusa Slicer doesn't have this issue on my system.

dstulken, 2024-04-18:
I can also confirm high CPU when switching to "Speed" in the preview.
MacBook Air M1, Orca Slicer 2.0.0

On the particular model on my plate, CPU usage with the "Line Type" view is using 19-25% CPU.   When I switch to "Speed", it jumps to 110% CPU.   The "Filament" view also is way up there (92% CPU), but others seem to be <25%.

CCS86, 2024-06-07:
@SoftFever can you reproduce?

Imploo, 2024-06-19:
Same here, preview tab slows laptop down even in other programs. 100% GPU usage, graph "Copy". I have a dedicated Nvidia GPU.

celulari, 2024-07-14:
Still happens on Orca 2.1.1

[image]
[image]

gthb96, 2024-07-17:
I can confirm the same, noticed it by feeling a warm breeze coming from under my desk in the morning

[image]
[image]

gthb96, 2024-10-08:
Has anyone tested if newer beta versions still have this problem?

Play13thStreet, 2025-01-17:
[image]

See this, please fix this, for my PC, it can handle it. Ryzen 7 5800X, 32GB RAM 3200, Radeon 7900XT, Watercooled CPU 74°C,
My Laptop, Ryzen 7 4800H, 16GB RAM, Nvidia RTX2060... is very laggy....

Felix14-v2, 2025-02-24:
Looks like a duplicate of https://github.com/SoftFever/OrcaSlicer/issues/3594 which is fixed in newer versions. Please, make sure that your software is up to date.

Orel-A, 2025-02-24:
Yes, this has been fixed here, as you can see here: https://github.com/SoftFever/OrcaSlicer/pull/6223

closed 2025-02-24 by Orel-A