OrcaSlicer/OrcaSlicer #4622: Max flowrate not overwriting filament setting?

WimVDK, 2024-03-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 beta

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambulab X1C

### How to reproduce

Using a third party upgraded hotend https://www.aliexpress.com/item/1005006700611159.html

1. Have a custom filament with flow set to 15 (like my Prusament ASA)
2. Run 2 tests (from 0-20 and 20-40) @ 265°C
3. Slice

### Actual results

That hotend is supposedly rated for 35mm³/sec max. Both tests completed perfectly with a standard non-cht nozzle. No underextrusion defects.

### Expected results

So by mf specifications defects should have started to appear at around the 30mm³/s mark. Yet it didn't despite slicing showing transitions and i could "hear" speed increases.

Either that hotend is more powerful or somethings "off" with the gcode sent. Currently upped my ASA filament max flow rate to 100mm³/sec which should be impossible to achieve even for an X1C with a 3rd party hotend. Loading a max flow rate test with matching steps (0-100). Results to follow

### Project file & Debug log uploads

[Max volumetric speed test.zip](https://github.com/SoftFever/OrcaSlicer/files/14705972/Max.volumetric.speed.test.zip)
kljlkml

### Checklist of files to include

- [ ] Log file
- [X] Project file

vgdh, 2024-03-21:
may be it a slow down printing setting in filament settings (cooling)

WimVDK, 2024-03-21:
[image]

Confirmed.

Max flow test does not overwrite custom filament setting. Attached pic is from top to bottom

- 20-50mm³ (filament setting flow 15mm³/s)
- 0-20 (ditto)
- 0-100 (max filament flow changed to 100 to allow higher flow). Defects on 265° @ just over 30mm/sec

otede, 2024-06-16:
I just did the max flow rate test on my A1. First defects spot on at 35 cubic mm per second. 
Orca 2.1.x beta

closed 2024-09-22 by github-actions[bot]

otede, 2024-10-13:
I think I'm facing a similar issue here. My understanding is that Filament profile Max Flow Rate should be the limiting factor Orca obeys, no matter what speeds I set. Otherwise it's pointlees. I've got a filament tested to around 35 cubic mm per second with no artifacts. But when I try to print it seems all is going too fast. I even cut down the max flow to 15 and I still get this

[image]