OrcaSlicer/OrcaSlicer #4625: Allow Vase mode limited to certain height range

Misterke, 2024-03-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I love designing and printing spiral vase mode stuff, but the slicer is imposing some limitations that aren't really needed.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Currently Spiral Vase mode is an all-or-nothing switch.  It is very useful to make strong boxes using curved walls with changing curvature over the height that adds strength, but the only thing you are still allowed to have apart from those walls is some bottom layers below the spiral path.  That significantly limits the cool stuff one could do with a bit more freedom and I do not believe that there's a reason for all those limitations.

Hence the request to allow for a `vase mode starting at a certain height` or even between z lower and upper limits.  Below the area where the vase mode then starts, one could have normal settings and apply modifiers.  One could ex. leave out the top and bottom layers to get an infill grid to show within the walls on which the spiral motion would then build.

And in principle you could even allow for normal printing to resume above / on top of the part printed in spiral mode.  That would then of course only be supported by the spiral walls, but with good design that could be enough.  It could ex. allow printing large sculptures in spiral mode while then moving back to normal printing to close the top off (using bridging) where overhangs would otherwise be too shallow for spiral mode to close the gaps, but where the bridging is limited enough to make it with only the (thicker) spiral walls.

Of course using this feature would be for people that know what they are doing, but for those it could open lots of options.

### Describe alternatives you've considered

I tried using spiral mode object with 5 x 0.2mm bottom layers and adding a modifier to change replace those bottom layers with infill, but this was refused.

ThinkBigPrintHuge, 2024-04-02:
+1
I can do this easily in S3D, but... I would rather not use S3D.

closed 2024-07-10 by github-actions[bot]

Misterke, 2024-07-21:
This request remains valid.  That nobody is attempting an implementation of it shouldn't be a reason to just close it ...

najwaa-debug, 2025-05-15:
This is a feature with great applications. Currently, I have to use S3D because it is the only slicer that offers this.