OrcaSlicer/OrcaSlicer #4628: Add newline between 'Filename format' and the text entry box

BETLOG, 2024-03-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The 'Filename format' text entry box is an impractical size.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add newline between 'Filename format' and the text entry box, so the text entry field can occupy the entire menu width.

### Additional context

lol
[image]

tlhintoq, 2024-03-25:
@BETLOG What's the size/resolution/magnification settings of your display?
I ask because it doesn't look like that on my my computer whether Orca is spread vertically across two monitors, or shrunk to half a monitor.  So I'm wondering if them is maybe something like a 4k laptop screen with text magnification at OS level set to 400% or some such combination.
[image]

BETLOG, 2024-03-25:
Its a 55" TV @1920x1080
But yours looks like it could benefit from my suggestion too. :)

tlhintoq, 2024-03-25:
I honestly am not sure what your suggestion means.
A `newline` is basically a paragraph ending.  Imperfectly described it would be a blank line of text.

Like that ^^^
But "between `Filename format` and the text entry box... That's horizontal space.   So I don't get it.

Do you mean between the check box above the text entry, and the text entry itself?
Or do you mean the text entry should have some internal padding so the text in the box isn't up against the border?
[image]

Space between the checkbox and textbox is probably doable.  Padding inside the textbox I don't _think_ is available.  I could be wrong but I think that's just a request from the program to the operating system:
`Hey, give me a text box here, and this is the text going in it` .  I don't know as internal padding is an option with the UI framework they use.  @SoftFever or @Noisyfox would be able to answer that.

BETLOG, 2024-03-26:
Make the filename field/box occupy the full width of the horizontal space UNDER the title: 'Filename Format'

tlhintoq, 2024-03-26:
> Make the filename field/box occupy the full width of the horizontal space UNDER the title: 'Filename Format'
> 
> Similar to the 'post processing scripts' field seen in [your screenshot](https://private-user-images.githubusercontent.com/11351506/316481871-fad902e8-cf6b-4440-89e1-71568801cfd0.png)

Ohhhhh...   Oh yeah, huh?  It would have to be its own section just like the scripts box.  The nature of the properties grid that its in is that its a left/right grid with field name on the left and control on the right: That's how it keeps its alignment.
But now that I get where you're going with that... it would make it easier to read and edit, no doubt.

closed 2024-07-02 by github-actions[bot]