OrcaSlicer/OrcaSlicer #4634: Extrusion rate smoothing is broken

vgdh, 2024-03-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

win 10

### Printer

kingroon kp3s

### How to reproduce

1. make cylindrical primitive 
2. enable Extrusion rate smoothing 

### Actual results

sudden change in flow where is no really change should be
[image]

### Expected results

smooth flow rate

### Project file & Debug log uploads

[proj and logs.zip](https://github.com/SoftFever/OrcaSlicer/files/14718576/prog.and.logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

vgdh, 2024-03-22:
how it looks in prusa
[image]

vgdh, 2024-03-22:
It's interfere with **small area flow compensation**
when I turned it off, extrusion flow smoother rate works well.

vgdh, 2024-03-22:
most likely it occurs because  small area compensation applied before flow smoother. And flow smoother process stumble upon very small very low flow path and do its thing.
@mjonuschat what do you think?

igiannakas, 2024-03-22:
> most likely it occurs because small area compensation applied before flow smoother. And flow smoother process stumble upon very small very low flow path and do its thing. @mjonuschat what do you think?

That is exactly it. The infill pattern creates small extrusions at the 90 degree mark of the large line segments where the nozzle changes direction. These are set to significantly less flow than the primary extrusion. Hence the extrusion rate smoothing feature kicks in and reduces the flow. 

Suggest you use monotonic lines as your solid infill of choice - it gives a better finish as these small extrusions can cause over extrusion and it also avoids this slowdown issue.

igiannakas, 2024-03-22:
The infill pattern creates small extrusions at the 90 degree mark of the large line segments where the nozzle changes direction. These are set to significantly less flow than the primary extrusion. Hence the extrusion rate smoothing feature kicks in and reduces the flow.

Suggest you use monotonic lines as your solid infill of choice - it gives a better finish as these small extrusions can cause over extrusion and it also avoids this slowdown issue.

vgdh, 2024-03-22:
> Suggest you use monotonic lines as your solid infill of choice - it gives a better finish as these small extrusions can cause over extrusion and it also avoids this slowdown issue.

Monotonic lines definitely helps
And it doesn't have print time penalty. I wonder why it's not a default infill?

igiannakas, 2024-03-22:
Think it’s just a carry over legacy from prusa and Bambu slicer.

SoftFever, 2024-03-23:
I got different result with the attached project(changed Top Surface pattern to Monotonic)
Although, I don't see the wrong doing
[image]

SoftFever, 2024-03-23:
From your screenshot, it seems fine.
It's indeed not necessary for the smoothing action in this case, but it's probably caused by the relatively small smoothing parameter(I noticed you set it to 10)?

[image]

vgdh, 2024-03-23:
yeah
When i switched to monotonic line the problem has gone.
So should I close the issue then?

Rickthebig, 2024-03-23:
> yeah When I had switched to monotonic line the problem has gone. So should I close the issue then?

No, I think the issue would still be there. If there are any incompatibilities when using flow smoothing with certain infill patterns, the software should at least prevent the user setting those patterns. It is no good idea to create a software that implies that you are acquainted with its quirks.

The clean solution would be to fix the bug allowing any infill pattern.

closed 2024-06-30 by github-actions[bot]