OrcaSlicer/OrcaSlicer #4635: Extrusion rate smoothing is wrong on infiil

vgdh, 2024-03-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

win 10

### Printer

kingroon kp3s

### How to reproduce

1.  make cylinder
2. enable extrusion rate smoother
3. gyroid infill

### Actual results

on some layers it's not smooth the extrusion rate on infiil start and on the others layer it does.
[image]
[image]

### Expected results

smooth extrusion rate at infill path start on every layer.

### Project file & Debug log uploads

[Cylinder.zip](https://github.com/SoftFever/OrcaSlicer/files/14719826/Cylinder.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-03-22:
It depends on the travel distance between the end of the last extrusion and the start of the next one. If it is over a few mm it will consider the pressure equalised naturally hence no need to run the algorithm as the nozzle is at 0 pressure.

vgdh, 2024-03-22:
> It depends on the travel distance between the end of the last extrusion and the start of the next one. If it is over a few mm it will consider the pressure equalised naturally hence no need to run the algorithm as the nozzle is at 0 pressure.
> 
> So this is a feature, not a bug.

So it will do zero to maximum pressure at infill start point. It not what pressure equalizer should do. 
In prusa slicer it works correctly and do it on every layer at any distance.

vgdh, 2024-03-22:
prusa sliced
every extrusion part at start has some pressure equalized path.
[image]

igiannakas, 2024-03-22:
It zeroes when:
1. There is a retraction
2. There is a travel above a certain minimum threshold

in both these cases there is no pressure in the nozzle so there is nothing to equalize. The prusa alpha release from a few days ago implements this as well.

igiannakas, 2024-03-22:
You can read more here: https://github.com/prusa3d/PrusaSlicer/pull/9622

vgdh, 2024-03-22:
> You can read more here: [prusa3d/PrusaSlicer#9622](https://github.com/prusa3d/PrusaSlicer/pull/9622)

Wow thank you for pointing that PR. Now it's clear why it's working that way!

> The main issue with existing code is not accounting for travel moves. You could travel move across whole build plate and pressure equalizer would still try to make a smooth transition from flow rates at each end. Extruder pressure drops to zero after a few mm of travel without extruding, and longer moves will retract on top of that. So it doesn't make sense to smooth flow between extrusions separated by a travel move more than a few mm.

vgdh, 2024-03-22:
Feature, not a bug.

closed 2024-03-22 by vgdh