OrcaSlicer/OrcaSlicer #4638: prime power position

loutwice, 2024-03-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

10

### Printer

Flsun v400

### How to reproduce

https://github.com/SoftFever/OrcaSlicer/assets/7931460/9e51cd87-8326-427f-9883-d54e4c98d4fd

### Actual results

cannot change the prime tower position

### Expected results

be able to move the prime tower 

### Project file & Debug log uploads

[Uploading joker test.zip…]()

### Checklist of files to include

- [ ] Log file
- [ ] Project file

WimVDK, 2024-03-22:
Maybe that's related to an issue i reported 2 months ago https://github.com/SoftFever/OrcaSlicer/issues/3776

loutwice, 2024-05-08:
bump @SoftFever

WimVDK, 2024-05-09:
don't think the tower size has anything to do with that, you can modify that to a degree under prime tower settings

loutwice, 2024-05-09:
@WimVDK  yup but same settings different behavior ...

loutwice, 2024-05-11:
still cannot relocate prime tower so i cannot print ... is there a parameter to play with?
[image]

closed 2024-08-18 by github-actions[bot]