OrcaSlicer/OrcaSlicer #4642: Convert single wall model to spiralizable multi-wall (strong) model

Rickthebig, 2024-03-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Spiral vase mode is wonderfull, as it allows high speed printing without defects caused by retractions and interruption of the extrusion path.

Problem: Models printed in spiral vase mode have to be single wall, and so are limited to the strength of one wall thickness. This is a limitation that can be overriden by the solution described here.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add an option in Orca to convert a single wall model to a multi-wall model that still can be printed in spiral vase mode (so this option "spiralizes" the model). This is possible by adapting the extrusion path to be a continuous line (not a loop) comprising all walls and sparse infill inbetween, not just the outmost one. This only can be done with models not containing isolated islands or layers usually composed of several loops. This option also implies certain constrains concerning the order in which the line types would be printed and the kind of infill pattern to be used.

A seam would still be normally visible at the sites where the line goes inwards from the outer wall, but it would be clean and less pronounced because there are no starts and stops of the line there, rather that is the site where lines hide going inside. This is still better than a weak one wall piece without any seam, because the model can be extruded without interruption nontheless.

To understand how this works, please take a look at this video:
https://www.youtube.com/watch?v=gCgyBmYq_bQ

Example models with several walls that can be printed in spiral vase mode:
https://www.thingiverse.com/thing:4201528

Options could be:

1. For single wall models: Increase the overall thickness by a given amount, and optionally include a special spiralizable infill pattern (limited patterns are possible then). The part will then be much stronger but still able to be printed in spiral vase mode. I don't see any fundamental obstacles preventing the implementation of this option, since the starting situation is unanbiguously defined. The vertical structures can be expanded inwards or outwards (let the user choose).
2.  For hollow models with several walls: spiralize the vertical structures by modifying the extrusion path to be continue (no loops). This demands an analysis of the model and I expect this to be more complicated than option 1 and not always possible. In this case, "spiralization" of the model does not render it stronger, but allows it to be printed using a continuous extrusion path.
3. Allow the user to define layer ranges that should be "spiralized", so avoiding possible limitations of the algorithms used in step 2, provided that those regions are recognized as "spiralizable" by the algorithm.
4. For solid models (not hollow), implementation would still be possible if adequate infill is chosen, but the algorithmic effort would be higher and the benefits might not be as evident.

### Additional context

While it can be argued that this "thickening" modification in case of one wall models is more a task for a designing software than for a slicer, by implementing this option the slicer is not breaking any canons. The slicer already allows itself to modify models by making overhangs printable, for example.

In case of multiple-wall or solid models, the implementation does not change any geometry but only changes the way the plastic is extruded and would be an acceptable choice in all cases where spiralizable infill patterns are strong enough for the use case.

This feature would let Orca shine above all other slicers, by the way.

Rickthebig, 2024-03-23:
Additionally, this new or enhanced way to implement spiral mode will need to make some constrictions void, like "only one roof layer" or no infill. Technically, the infill in this enhanced spiral mode is part of the outer wall or its continuation inwards.

closed 2024-06-30 by github-actions[bot]

j0e57, 2025-02-14:
@Rickthebig If you still interested can check my new OpenSCAD parametric double wall, for basic shape.. better than waiting it implemented on slicer :)
https://www.thingiverse.com/thing:6928805