OrcaSlicer/OrcaSlicer #4643: Request for Flashforge Dreamer / Dreamer NX printer profiles

atanisoft, 2024-03-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I've not been able to locate a profile for the FF Dreamer or FF Dreamer NX printers. I'd like to use my FF Dreamer printer with it's dual nozzles via OrcaSlicer.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Addition of profiles for Flashforge Dreamer and Dreamer NX printers

shafror, 2024-04-17:
I second this. 

I'm told that my Dremel 3D45 has the same firmware as FF Dreamer NX, and would love to get away from using outdated Cura that works with Dremel.

a-dam-berger, 2024-05-02:
I agree. There really isn't a good slicer for Flashforge Dreamer as the Flashprint software is junk compared to all others and I would love to switch to Orcaslicer as I also have a Bambu X1C in the shop

atanisoft, 2024-05-06:
I wouldn't say that FlashPrint is junk entirely, it just leaves a lot to be desired.

snintendog, 2024-08-04:
Would Third this If i remember right its a minor change from the Dremel G Code and there was an oddity with the dual Nozzle Dreamers in Cura that only the left nozzle works.

Bretterson, 2024-09-15:
Fourthing this!

X4merxb, 2024-09-28:
I too would like to have these settings for the Flashforge Dreamer.  Thank you

atanisoft, 2024-12-30:
This support would still be beneficial for end users, though I've stopped printing with my Dreamer in favor of the AD5MPro (and hopefully soon AD5X)

X4merxb, 2024-12-30:
I still have it and still use the dreamer even though I have a 5m pro I
just think it would be awesome to be able to control them both from Orca
slicer

atanisoft, 2024-12-30:
Yeah, I'd use my Dreamer as well but haven't had a specific need to recently. But I'd still use it if it was supported here rather than needing to figure up FlashPrint to talk to it

X4merxb, 2024-12-30:
Exactly

silvandre, 2025-03-11:
Hi Guys! any news on this matter?

X3R0theH3R0, 2025-06-27:
I fifth it.

3d-mac, 2025-09-09:
Any updates? As I'm also keen! 
I tried to get FF Dreamer profile working with other slicers myself, but got stuck.