OrcaSlicer/OrcaSlicer #4648: Odd bridging in i.e. Z of the orca cube

Arakon, 2024-03-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 beta

### Operating System (OS)

Windows

### OS Version

Win 11

### Printer

VZBot330

### How to reproduce

- Load the attached project
- Slice
- Notice that the top infill of the "Z" is actually bridging and oddly broken up into wall lines

### Actual results

[image]

### Expected results

Surface should be smooth, and used to be in 1.9.0 (if I remember correctly)

### Project file & Debug log uploads

[OrcaCube.zip](https://github.com/SoftFever/OrcaSlicer/files/14726710/OrcaCube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2024-03-23:
Thank for you reporting.
Fixed

Felix14-v2, 2026-10-07:
I'm unable to reproduce the issue with the provided 3MF:

[image]

PR #4651 (2024-03-23): Wrong bridge detection result
Fixed a regression bug that top surface was mistaken as bridge.
To produce this bug, just set thick top shell layer when slicing Orca cube
[image]

Fixes #4648

@igiannakas 
I reverted this part of the change. I tested the case you mentioned, it seems fine too
https://github.com/SoftFever/OrcaSlicer/pull/3285#discussion_r1444632150
closed 2026-10-07 by ianalexis