OrcaSlicer/OrcaSlicer #4655: Custom names for the bed type plates.

vgdh, 2024-03-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Here's the corrected version of your text:

"Would it be cool to be able to use custom-named plate types? 
I personally want only two plates: Glass and PEI. 
The rest of the plates are useless for me personally. 

I think some of us have our own plates and may have more and different types like:
1. Spring Steel Sheet with PEI
2. Glass Build Plates
3. Polypropylene (PP)
4. Garolite Build Plates

Some even have several different glass plates and different bed mesh profiles for them in Klipper. 
If you can choose custom names, you can use them in Klipper macros. Automation and making it easier to print is good."

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

The structure may be inherit like.

- **Glass** (it would appear at all profiles as a plate name)
|---- **favorite glass** (doesn't apper at profiles but this will be sent to klipper as placeholder)
|---- **rough glass** (doesn't apper at profiles but this will be sent to klipper as placeholder)
- **PEI** (it would appear at all profiles as a plate name)
|---- (name isn't chosen so 'PEI' will be sent to klipper as placeholder)

TexZeTech, 2024-03-31:
I too would like to see this added.

driest, 2024-04-05:
As an interim solution I'd already be happy to rename the 4 pre-existing ones. That helps prevent mistakes when building custom g-code, as it's easy to forget that and then scrape your bed when you put in the actual "Engineering Plate", after using that slot for a POM plate with custom z-offset, for example.

vgdh, 2024-07-05:
up

closed 2024-10-11 by github-actions[bot]

AJolly, 2024-10-20:
This would be great

RoBoT24-web, 2024-11-16:
This would be great

GabrielMASU, 2025-02-27:
UP,UP,UP

masterjuggler, 2025-03-13:
Are there any plans to implement this? I also have several different bed types that have nothing to do with bambu lab, and need to cross reference the names on a chart.

DoctorWiz, 2025-05-12:
Ditto.
I have 6 different types of build plates.  Three of them correspond to the four predefined types, although I'm not happy with how they are named.  That leaves one useless predefined type I never use, and 3 more plate types I can't select (cuz they don't exist).

Can you at least tell me what .json file they are defined it, so I can edit it?

DoctorWiz, 2025-05-14:
With the utmost gratitude and respect for all the wonderful developers that volunteer their time to this great project...   
... I also think this is a needed feature.  Build plates and their coatings is still an area of active research and development and new types are released often (admittedly, not all of them great).  And each printer manufacturer seems to have their own names for them.
I personally have: Smooth PEI, lightly textured PEI, heavily textured PEI, Epoxy, Cold, FR4/Garolite, Holographic (multiples), and even Glass.  And there are several sub-types for some of those.

Rodamyot, 2025-07-19:
I would also love to have this feature. Personally, I have 4 types of texture in my plates and each requires a different Z offset depending on the type of filament.

3LD, 2026-11-16:
'Support Multi Bed Types' printer setting only useful if you have a variety of build plates. 
'Support Multi Bed Types' printer setting useless if you have too many build plates. 

'Engineering Plate', 'Smooth Cool Plate', etc  is too broad. I've got multiple plates that fall under most of the hardcoded categories that each require different settings.