OrcaSlicer/OrcaSlicer #4660: Add an override for Z Offset in filament settings

Rickthebig, 2024-03-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

If changing filaments,  the Z Offset might need to be different because it very much depends on the filament material.
Currently, Z Offset is in the printer profile and aims to compensate for bad distance between nozzle and bed. This ignores the fact that the Z Offset often needs to be adapted to the filament material used, regardless of printer conditions.

For example, if my printer has been well calibrated concerning height of the build bed and I mostly print in PETG, everything is fine. But if I change the filament to PLA, I need the nozzle closer to the bed for the 1st layer to adhere properly (PLA needs more "squish"). Printing PLA using the same printer profile could fail, although it is the same printer.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

In order to account for filament needs, the Z Offset should be put in the override list of the filament profile. In consequence of doing this, the user would change the filament used to print without having to take care about changing printer settings (which would be not logical since he is using the same unchanged printer).

I have had print fails because I forgot to adapt the Z Offset when changing the filament material.

### Describe alternatives you've considered

Alternatively, define two different Z Offsets: "Printer Z Offset" for the printer, valid for all filaments and used to remedy bad height calibration, and "1st Layer Z Offset" for the filament, which would be saved in the filament profile. In this case of having two different Z Offset types, the Z Offset value for the filament concerns only the 1st layer, meaning that the second layer would need to compensate the height difference introduced, to avoid an overall change of the printed piece dimensions.

### Additional context

See also closed (orphaned) request:
Custom Overrides #2869

streetgt, 2024-03-23:
If you are using Klipper you can do it easy... I don't see any advantage or doing it tbh.

Luro02, 2024-04-26:
> If you are using Klipper you can do it easily... I don't see any advantage or doing it tbh.

Not everyone is using Klipper.

bmorgan99, 2024-05-02:
When I put g-code in the filament settings, Advanced, Filament Start G-code box to raise the z axis, it does so for the model, but the skirt still gets smashed to the same degree regardless of the Zx number I enter.  Shouldn't it apply to the skirt as well?

rmeissn, 2024-06-15:
I'm on Marlin2, thus can't use the Klipper macro.
My solution is borrowed from the cura z-offset plugin. Move the printhead somewhere where no filament was extruded yet and execute in the filament start gcode:
```gcode
G0 F6000 Z0.15 ; Adjust Z by -0.05 when initial layer height is 0.2
G92 Z0.2 ; Save as Z 0.2
```
All subsequent moves to e.g. Z0.4 will actually resolve to a move to Z0.35, even though 0.4 is written in the gcode

klhutchins1, 2024-07-02:
I’d like to request the addition of a feature that allows for easy switching between different Z-offset values for various filament types, such as PETG and PLA. This would be particularly useful as these materials require significantly different Z-offset settings to achieve proper adhesion (e.g., a Z-offset of .09 for PETG and .04 for PLA). TPU also has unique requirements in this regard.

Creating separate printer profiles solely for Z-offset differences is not efficient. It introduces complexity and the potential for errors when tuning other settings, as adjustments would need to be duplicated across multiple profiles. Additionally, adding custom G-code for Z-offset changes is not always clear and can complicate troubleshooting and future tuning efforts.

We already have filament profiles to manage vendor and material differences, which helps keep settings organized. Adding a Z-offset override in these profiles would be consistent with existing organizational structures and just as crucial as other settings like retraction.

This feature would enhance usability and streamline the process of switching between filament types, ensuring optimal print quality without the need for redundant profiles or opaque G-code adjustments.

Thank you for considering this enhancement. I believe it would greatly benefit users who frequently switch between different materials.

hselomein, 2024-09-21:
I agree this would be useful, as of now I'm trying to write a condition statement that uses different z-offset depending on filament type and if I a using a 0.2 or 0.3mm first layer/

d3dorsett, 2024-12-13:
I would agree this would be helpful. My Elegoo PLA+ is 0.2 vs all my other PLA at 0.3. Having an extra printer setup for this one filament is a bit inconvenient. PETG is 0.35, so now I have 3 printer profiles and 3 filament profiles.  If the change is made, thank you.

[image]

DrKropp, 2025-06-11:
I would also love this feature. I regularly print PLA, ABS, and ASA with Z_Offset of 0, PETG with 0.030, PET_CF at 0.0150, and PPS-CF at 0.020. If I forget to change it when I start a print with PET-CF it will rip the PEI right off the textured build plate. I could put some values into Klipper and use conditionals, but these values might vary between filament manufacturers, and I am not comfortable enough with Klipper code to catch all of the use cases.