OrcaSlicer/OrcaSlicer #4664: Orca Slicer G-code not printing on Neptune 4 plus

yprater, 2024-03-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

macOS

### OS Version

MacOS Ventura 13.6

### Additional system information

Model Name: MacBook Pro
Chip: Apple M2 Pro
Total Number of Cores: 12 (8 performance and 4 efficiency)
Memory:	32 GB
 Memory Type: LPDDR5
System Firmware Version: 10151.1.1
OS Loader Version:	8422.141.2

### Printer

Elegoo Neptune 4 Plus

### How to reproduce

1. Load model/STL file in Orca Slicer
2. Slice plate
3. Export G-code file to USB
4. Plug USB into printer and select select G-code to start printing

### Actual results

Below is an image of what I'm seeing when actually trying to print the model. You can see it says printing and but the progress says 0hrs and 0 minutes and nothing actually happens.

[image]

### Expected results

The printer should be printing.

### Project file & Debug log uploads

[debug_Thu_Mar_21_16_28_46_54155.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14733142/debug_Thu_Mar_21_16_28_46_54155.log.0.zip) - Log file

[EN3P_charmander.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14733146/EN3P_charmander.3mf.zip) - project file

[Bulbasaur_0.2mm_PLA_Elegoo Neptune 4 Plus_8h12m 1.gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/14733148/Bulbasaur_0.2mm_PLA_Elegoo.Neptune.4.Plus_8h12m.1.gcode.zip) - G-Code

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2024-03-24:
I don't see anything wrong in the gcode file.
What error did Klipper show?

yprater, 2024-03-24:
> I don't see anything wrong in the gcode file. What error did Klipper show?

What is Klipper? I’m not familiar.

No error is being thrown in Orca itself. It’s slicing, the g-code is exportable but when I actually pop the drive into the printer and select the g-code to print, nothing happens.

yprater, 2024-03-27:
@SoftFever  - thoughts here?

SoftFever, 2024-03-28:
Sorry I can't help here.
It's better to contact their customer support.

yprater, 2024-03-28:
I emailed their Support two weeks ago and have heard nothing. I DM’d them on Instagram, which told me to go to Discord, posted there and was told to open a bug here.

Do you know their contact info? support@obico.io  hasn’t yielded a response

closed 2024-03-28 by yprater

josepatron, 2024-05-22:
It happens to me with a DIY printer but it moves all the Start Gcode instructions, when temps are ready and everything pósitiones..... ????? it does not print.