OrcaSlicer/OrcaSlicer #4669: Please add profile for Creality Ender 3 S1 Plus

Gozania, 2024-03-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The entire S1 line has been added except for the often left out S1 Plus.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The entire S1 line has been added except for the often left out S1 Plus.

### Describe alternatives you've considered

1. I have modified a S1 Pro profile to have a print size of 300x300x300.

cochcoder, 2024-03-24:
How well does the profile work when you edit the S1 Pro profile to have a print size of 300x300x300? Are there any values that should be changed for speed, acceleration, retraction, etc?

vgdh, 2024-03-24:
If the mechanical side of the printers is the same. You should be ok.

cochcoder, 2024-03-24:
@vgdh Let me know if PR #4674 works for you, let me know if you need a download link

Gozania, 2024-03-24:
> How well does the profile work when you edit the S1 Pro profile to have a print size of 300x300x300? Are there any values that should be changed for speed, acceleration, retraction, etc?

I havent printed anything with orca yet.  Still in the process of moving print profiles & filaments over to Orca from Prusa.  I know there is a script that does this, but I'm hesitant due to my lack of CLI skills.  If something goes wrong, I wont know how to recover.

cochcoder, 2024-03-24:
What is the script called/located? Also if you can run a test print with the PR I created to make sure that everything works, that would be awesome!

Gozania, 2024-03-24:
> @vgdh @Gozania Let me know if PR #4674 works for you, let me know if you need a download link

I just tried to import the creality.json file, but it did not take.
[image]

[image]

Gozania, 2024-03-24:
https://github.com/SoftFever/OrcaSlicer/discussions/1636

I wonder why softfever decided on json rather than .ini which is what Prusa uses...

cochcoder, 2024-03-24:
> I just tried to import the creality.json file, but it did not take.

That won't work as there are 14 other files that were added/changed, here are a few download links for what main would be once merged: [Windows portable](https://github.com/cochcoder/OrcaSlicer/actions/runs/8407801159/artifacts/1353391984), [Linux Appimage](https://github.com/cochcoder/OrcaSlicer/actions/runs/8407801159/artifacts/1353390677), [Mac Arm](https://github.com/cochcoder/OrcaSlicer/actions/runs/8407801159/artifacts/1353373818), and [Mac x64](https://github.com/cochcoder/OrcaSlicer/actions/runs/8407801159/artifacts/1353396764)

Gozania, 2024-03-24:
Not quite I understand what ya mean by "once main would be once merged"  Please forgive the ignorance, I'm just a roughneck who enjoys CNC & lasers....  Do you mean if I am on the beta version instead of 1.9.1 my printer will be there?  I poked around in the windows mobile zip & found the S1 Plus 0.4 Json and tried to import that, but still no dice.  Same error window as before.

I do appreciate ya trying to help me out.

cochcoder, 2024-03-24:
Because of the way that the program is setup I am 98% sure that you can't copy the .json file as you would have to also compile the software. What you need to do is to extract the Windows portable.zip and run orcaslicer.exe, which should show the Ender-3 S1 Plus. I don't recommend using this version of OrcaSlicer on a day to day biases due to still being developed. You'll see the Ender-3 S1 Plus in the stable release once that comes out and my PR has been merged.

Gozania, 2024-03-24:
Gotcha, so if I wanted to see the printer now, I would just run the portable app through a "container" (forgive my incorrect verbage), until the stable release comes out.

I think I will just wait a while until things are ironed out & there is also a native way to simply import settings from Prusa directly into orca.  I tried that script I mentioned with zero success.  All my installations are in default locations, so dont understand nor skilled enough to know why it failed.

Will continue to use prusa until I am really bored & actually have time for the monotonous task of transferring all my settings manually to orca.

Hopefully softfever can find some sort of a solution to transfer settings via a few simple clicks for non-cli/git boneheads like myself.

Again, thank you for all your help!

cochcoder, 2024-03-24:
> Gotcha, so if I wanted to see the printer now, I would just run the portable app through a "container"

Yes, that is correct

> Will continue to use prusa

That is fine, however if you or @vgdh can do a test print with this new profile that would be greatly appreciated so that I can fix any simple issues that might arise when you or someone else uses this profile.

cochcoder, 2024-03-24:
I changed a few things: [Windows portable](https://github.com/cochcoder/OrcaSlicer/actions/runs/8411678737/artifacts/1353945086)

Gozania, 2024-03-24:
For whatever reason it was zipped inside of a zip, but I did get the latest version you sent working.  I'm moving over my "fast" profile from prusa to orca manually, will slice a benchy and let ya know how it turns out.

Side note... I only checked the box for .4 nozzle, but it added all the nozzle sizes.  Also the bed is model & texture is off.  It is looking like a standard S1/E3v2 rather than a S1 Plus.  The bed on the S1 Plus is exactly like this & is PC than PEI  https://www.amazon.com/Ender-Frosted-Magnetic-Flexible-310x315mm/dp/B0B8HDRSYF/ref=sr_1_1?crid=1XEAG8YC8NGSW&dib=eyJ2IjoiMSJ9.mXQZtPbjqthoaGM6ok7ZoAxkmKsWXODYOe_bMxPeWLuDKfmu_9xMQmwTyM1FeBJpPcpfJ-XkCAkvSnRveOjrkkxV4CeYUJQ3BadtZ1jo0xMa3BYtiaoAPm0ZylyJyUmlTXIQSZXMvC4MEEONhuaKih-5t8ifR3eXaaKx8X_6RCuY-bcIm2oucr6AVd1L3mpjTvJd0Rsd1FKRIrFA4sdrmTtULWQxX5PI3ajJrfAus-c.SYd5tWdRJD-lODMaXqLhtuFgTtEQg7Xh0cxEh-xr5_w&dib_tag=se&keywords=s1+plus+bed&qid=1711324329&sprefix=s1+plus+be%2Caps%2C176&sr=8-1

It would be even better if you a 300x300 border can be added outside the bed grid to show limits.

cochcoder, 2024-03-25:
> Side note... I only checked the box for .4 nozzle, but it added all the nozzle sizes.  Also the bed is model & texture is off.  It is looking like a standard S1/E3v2 rather than a S1 Plus.  The bed [...]

Thx, I'll look into it, I think I fixed the issue with all four nozzles being added with only one chosen. 

> It would be even better if you a 300x300 border can be added outside the bed grid to show limits.

Do you mean fixing the bed model/texture?

Gozania, 2024-03-25:
Yes the bed model/texture.  Prusa used to have an exact replica, but dissapeared somehow.   I made my own less desirable one using a CR10 model & ender 3 texture.  Does the job, but Prusa's initial one was much better.

cochcoder, 2024-03-25:
> Yes the bed model/texture. Prusa used to have an exact replica, but dissapeared somehow. I made my own less desirable one using a CR10 model & ender 3 texture. Does the job, but Prusa's initial one was much better.

How does the Ender-3 V3 SE bed model/texture look? 
[image]

Gozania, 2024-03-25:
LOVE IT!

I'm having trouble navigating.  I am searching for how to modify the output of my gcode.  Also is there a way I can send the gcode to my printer directly into a specific directory but not print yet?  This was easily done through PS.

Screenshot_20240324_203639.png (view on web)<https://github.com/SoftFever/OrcaSlicer/assets/103969142/f6c28f70-9618-4faf-a3fb-ef86c777c7fe>

Gozania, 2024-03-25:
Looking for this....

Gozania, 2024-03-25:
FOUND IT!

Gozania, 2024-03-25:
Now this is the last thing I am after before I send it to be printed....

cochcoder, 2024-03-25:
I am currently compiling a new version of the software and hopefully, it fixes all the issues you found so far

Gozania, 2024-03-25:
I found the output section and I see the upload section just like PS...  However I did have an issue with just copying my PS output string to Orca.  It didnt work and thats howcome I was pestering about the directory.   The output string I use in PS is as follows.

{input_filename_base}_{print_time}_${total_cost}_{total_weight}g_{temperature[0]}C_{filament_type[0]}_{printer_model}.gcode

Where did my original string go wrong in orca?

cochcoder, 2024-03-25:
> Where did my original string go wrong in orca?

It may be the ".gcode" string at the end, I don't know much when it comes to that area of the slicer, I recommend asking around if you don't find a solution. Also, the Bed Mesh leveling may not happen when you start a print as I just found that issue and it takes around 40 minutes for the code to recompile.

Gozania, 2024-03-25:
TY for letting me know that, allthough I use BED_MESH_PROFILE LOAD=default in my start gcode.

First print didnt stick and failed right away.  I also noticed movement is much slower compared to the settings I have in PS which I copied over.

Gozania, 2024-03-25:
One thing that would be MASSIVELY helpful is a search function like PS.  It comes in soooo handy to find the setting you want.

cochcoder, 2024-03-25:
> One thing that would be MASSIVELY helpful is a search function like PS. It comes in soooo handy to find the setting you want.

Orca Slicer does have a search function, although it is broken up into three different setting locations. I'll keep it in mind though, might add an overall search function if I get time ;)

Gozania, 2024-03-25:
Yeah a simple magnifying glass in the tool bar, totally sufficient.

I think I found my speed issues...  It seems you have to have a higher speed rate in machine limits rather than just telling the slicer to ignore them.

Reslicing, re-doing z-offset and will see what happens.

cochcoder, 2024-03-25:
> First print didnt stick and failed right away.  I also noticed movement is much slower compared to the settings I have in PS which I copied over. 

If you have any changes to suggest making to the profile, please let me know! (As long as they are not too exorbitant (the majority would be fine with them))

Gozania, 2024-03-25:
I should have mentioned it earlier.... I'm running it on klipper and a sonic pad, since I purchased it a year ago, my settings wouldnt be ideal for a stock deployment.

cochcoder, 2024-03-25:
> I think I found my speed issues... It seems you have to have a higher speed rate in machine limits rather than just telling the slicer to ignore them. 

I'll look into it when I can as I just copied the S1 Pro's profiles 

> I should have mentioned it earlier.... I'm running it on klipper and a sonic pad, since I purchased it a year ago, my settings wouldnt be ideal for a stock deployment.

Thank you for mentioning this! This does limit the settings that you can reliably give feedback on, however, speed and acceleration are on of the categories that you can give feedback on for the stock deployment

Gozania, 2024-03-25:
I can surely do that!  The profiles I use are the same slightly modified ones that creality provided with the initial launch of the sonic pad.  They are pretty much stock other than changing layer heights UP .01 to make it an even number.  I also swapped the accelertations between their normal and fast profiles as it looked like they had them swapped... A slower accel was present on the faster profile, i.e. 1500 accel was on the fast and 2500 accel was on the normal.

I have attached a copy of creality's profiles for the S1 Line & Ender3v2.  Please keep in mind the accel swapping I had to do earlier when implementing them....

Gozania, 2024-03-25:
One thing I strongly dislike is my maximum acceleration for travel in machine limits is capped to 1000 mms.  I have no issues with this being 5k mms in PS.

cochcoder, 2024-03-25:
> I have attached a copy of creality's profiles for the S1 Line & Ender3v2

This copy didn't come through for whatever reason, also I just finished compiling the newest [version](https://github.com/cochcoder/OrcaSlicer/actions/runs/8413560744/artifacts/1354256341)

Gozania, 2024-03-25:
odd...  take 2

Another thing...  I love the way Orcas thumbnails look over PS, however it would be nice if the thumbnail was the same color as selected for the filament.   Or is there a way to customize this for each print before uploaded?

cochcoder, 2024-03-25:
> it would be nice if the thumbnail was the same color as selected for the filament. Or is there a way to customize this for each print before uploaded?

I haven't seen a way for this to be customized, however, I don't use thumbnails so take it with a hundred-pound bag of salt. You may want to look around for a existing issue or create a new issue for this

Gozania, 2024-03-25:
Copy....

I have a benchy going with my speed settings for klipper.  If youd like I can try to print one just using a stock profile provided.

cochcoder, 2024-03-25:
> Copy....
> 
> 
> 
> I have a benchy going with my speed settings for klipper.  If youd like I can try to print one just using a stock profile provided.

I'd like to make a few more changes to the profile tomorrow, before you do a test print with stock settings

Gozania, 2024-03-25:
Roger that...

Benchy is looking pretty good.  However I do see my accell & deccel is being capped by orca.  I dont do anything crazy, the max I go is 5K accel & 2500 for deccel.  Orca is capping me at 2.5k accel & 750 decell.  Is there a way to overcome this?  My machine runs the latter no problems at all.

cochcoder, 2024-03-25:
> Orca is capping me at 2.5k accel & 750 decell. Is there a way to overcome this?

In the printer's settings (printer profile) you can edit the max motion values, like acceleration. Then in the print profiles (Draft, Optimal, etc) you can change those motion values to whatever you want, as long as it is below the printer's max

cochcoder, 2024-03-25:
> My machine runs the latter no problems at all.

Great! Is the issue with the bed model/texture fixed as well?

Gozania, 2024-03-25:
Oh I havent used the newst compilation.  I am running on what ya initially gave me when I said it was zipped within a zip.

cochcoder, 2024-03-25:
Sorry I got a bit confused

Gozania, 2024-03-25:
No worries!
      In the printer's settings (printer profile) you can edit the max motion values, like acceleration. Then in the print profiles (Draft, Optimal, etc) you can change those motion values to whatever you want, as long as it is below the printer's max

I have everything edited the way I want EXCEPT for travel, which it wont let me edit.
I would love travel to be 5K which is what I travel at.

Gozania, 2024-03-25:
First orca benchy came out looking great!

Gozania, 2024-03-25:
So new discovery... My PS Fast profile uses The same printing accell & decel that orca was showing me.  However I am indeed being capped by Orca ONLY for my travel speed which I would like to be 5K (I have no issues) as I do in PS.  The last screen shot I showed you has only travel speed greyed out & un-editable.

Gozania, 2024-03-25:
Can you send me a link to the latest version you have compiled?  I tried to export my configs and the program crashes.

I however was succesfull importing my filaments into Orca via that script.

Gozania, 2024-03-25:
After running that perl script, all my filament colors from PS were added into Orca & the thumbnails now reflect the assigned color!

cochcoder, 2024-03-25:
Here is the new [version](https://github.com/cochcoder/OrcaSlicer/actions/runs/8424675278/artifacts/1356791088), I changed the max travel acceleration to 5,000 and the default print profiles should be ready for testing!

Gozania, 2024-03-25:
TY!  I'm printing some parts at the moment, so machine is tied up.  I tried adaptive mesh, probed fine but nozzle dragged across bed after purge I know adaptive purge is available, but not sure if I can get it on Sonic Pad as crapality likes to lock down open source software for their own use.

After I am done with these parts (will take a while) I will delete the old version and download/extract this new build and give it a go.  Side note, I am not getting the modified bed yet after installing a different version you provided.

Is it possible to add new beds?  Or does the bed get reset during every project.  Ideally I would like to have my S1 Plus be smooth PC/PEI & My Ender3v2 have a Textured PEI & Smooth PEI each saved & named.  Forgive the ignorance still learning how to navigate.  I also found the search tool in the print settings section.

cochcoder, 2024-03-25:
> Is it possible to add new beds?  Or does the bed get reset during every project.  

Do you mean this? 
[image]

> Ideally I would like to have my S1 Plus be smooth PC/PEI & My Ender3v2 have a Textured PEI & Smooth PEI each saved & named.

Do you mean that you want different bed textures? I'm not sure what you're asking for. 

> Forgive the ignorance still learning how to navigate.

Don't worry about it! If you need a visual guide, [Teaching Tech](https://youtu.be/cquTCpz1V74?si=RpU6gzZ-OAsrWugd) made a fantastic video walking through Orca Slicer and its advance features

Gozania, 2024-03-25:
YES!
      "Do you mean this?
      Screenshot_20240325_171303.png (view on web)<https://github.com/SoftFever/OrcaSlicer/assets/103969142/adcfbb8a-21a1-412a-a08a-a708e6a5016a>"

Yes, only for one of my machines since it has a dual sided bed and I do use both sides.
      "Do you mean that you want different bed textures? I'm not sure what you're asking for."

TY for that!  Gonna give that a watch while this print is going!
      "Don't worry about it! If you need a visual guide, Teaching Tech<https://youtu.be/cquTCpz1V74?si=RpU6gzZ-OAsrWugd> made a fantastic video walking through Orca Slicer and its advance features" 

Screenshot_20240325_171303.png (view on web)<https://github.com/SoftFever/OrcaSlicer/assets/103969142/adcfbb8a-21a1-412a-a08a-a708e6a5016a>

cochcoder, 2024-03-25:
@Gozania  Just letting you know that I am making a small change in the profiles that could greatly reduce print times (depending on the model size/infill needed), so you may want to wait 40 minutes before doing another test print if you were about to do one

Gozania, 2024-03-25:
LOL!!!!   TY for telling me!  I just installed the latest you provided & tested out the export function before removing my S1 Plus entirely & it still crashes right after I click export configs.

cochcoder, 2024-03-25:
> Yes, only for one of my machines since it has a dual sided bed and I do use both sides.

If you go into your Printer's settings you can enable multiple bed types and then you can change the bed type on the main page, hope this helps!

> I just installed the latest you provided & tested out the export function before removing my S1 Plus entirely

Talk about timing! 

> it still crashes right after I click export configs

Weird, I'll keep a eye on that

Gozania, 2024-03-25:
I'll try this with your next compile

      If you go into your Printer's settings you can enable multiple bed types and then you can change the bed type on the main page, hope this helps!

Side note, would be awesome if you added a couple more common filament manufacturer profiles i.e. Proto Pasta  & Hatchbox offerings.

cochcoder, 2024-03-25:
> Side note, would be awesome if you added a couple more common filament manufacturer profiles i.e. Proto Pasta  & Hatchbox offerings.

Make a new issue if there isn't one, link it here, and I'll look into it when I get time

Gozania, 2024-03-25:
Will do!  Again, TY for that video.  I now understand how the plates work.

cochcoder, 2024-03-25:
> Again, TY for that video.  I now understand how the plates work.

No problem, it is my pleasure!

cochcoder, 2024-03-26:
Thank you for waiting, the new download with combine infill enabled by default and added draft profiles is [here](https://github.com/cochcoder/OrcaSlicer/actions/runs/8428192242/artifacts/1357594305). Please let me know what the results are and if you know of any changes that should be made to the default profiles. Thank you again for your testing!

[56 more replies]

PR #4674 (2024-03-29): Add Creality Ender-3 S1 Plus
Closes  #4669, closes #3707
closed 2024-03-29 by SoftFever